OpenIntelWireless/itlwm #6: 关于一处编译bug和非10.15的问题

Cao-czy, 2020-03-17:
在 mac80211.cpp 中 2174 行
`
IOReturn itlwm::
_iwm_start_task(OSObject *target, void *arg0, void *arg1, void *arg2, void *arg3)
`
用到了 itlwm.hpp 中 的 _iwm_start_task 
`static IOReturn _iwm_start_task(OSObject *target, void *arg0, void *arg1, void *arg2, void *arg3);
`
但是在编译 OpenWifi 的时候并不会编译 itlwm.hpp 而是用了 OpenWifi.hpp ，我在 OpenWifi.hpp 中把 _iwm_start_task 放过去了就好了，不知道是不是我的问题。

编译成功后 -v 中显示 IO80211Controller 找不到，我在代码中发现
[image]
只提供了10.15的内容，不知道是不是通用的呢

MarshalX, 2020-03-17:
https://github.com/zxystd/itlwm/pull/4

johnnync13, 2020-03-17:
@zxystd Will you no longer work on the other repository? https://github.com/zxystd/AppleIntelWifiAdapter

zxystd, 2020-03-18:
I will abandon IO80211Family and self write a network management client, and implement Airdrop function further, so this is not need to fix.

zxystd, 2020-03-18:
> @zxystd Will you no longer work on the other repository? https://github.com/zxystd/AppleIntelWifiAdapter

just paused.

closed 2020-03-19 by zxystd