OpenIntelWireless/itlwm #8: Stupid Question

0x47726567, 2020-03-18:
What is the difference between this project ( itlwm ) and  AppleIntelWifiAdapter?

williambj1, 2020-03-18:
I think this is already explained in README

deniro98, 2020-03-18:
> What is the difference between this project ( itlwm ) and AppleIntelWifiAdapter?

This is S/L/E Kext, work with open wifi.

williambj1, 2020-03-18:
> > What is the difference between this project ( itlwm ) and AppleIntelWifiAdapter?
> 
> itlwm is S/L/E Kext, work with open wifi.

You'd better stop joking.

0x47726567, 2020-03-18:
Welp, i guess i'm blind.

williambj1, 2020-03-18:
> Because the integrated 80211 ported from OpenBSD in that project has so many things require testing, I came up with an idea of porting the entire iwm driver from OpenBSD, and this repo is the result.

I thought I'm blind as well.

closed 2020-03-19 by zxystd