OpenIntelWireless/itlwm #9: Kernel panic after loading kext and using modem

MarshalX, 2020-03-18:
Modem: **Huawei e3372h-153**

Maybe this is not important at all, but I noticed such a feature.

Steps:
1. load kext 
2. *auto*connect to wifi, using eithernet
3. disable wifi hotspot
4. plug in modem in usb
....
kernel panic

As I remember same issue occurred with https://joshuawise.com/horndis (eithernet from phone by usb)

vasishath, 2020-03-18:
Omg I use horndis kext and I faced the same kernel panic

zxystd, 2020-03-19:
So many things I haven't handled, but this stage these things are not very important rather than smooth stable connect to the internet and hardware compatibility, I will record it. Thanks.

zxystd, 2020-05-29:
Maybe fix it isn't necessary because this situation is rare...

closed 2020-05-29 by zxystd