OpenIntelWireless/itlwm #10: AC 9462卡在tsleep_nsec iwmfw

htmambo, 2020-03-19:
查了下，9462与9560用的固件应该是相同的，就自己在代码里加了下9462的product_id(0x02f0)编译了下，加载时卡在tsleep_nsec iwmfw了
2020-03-20 02:31:15.294024+0800 0x271      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW callback ret=0x00000000 length=2678284
2020-03-20 02:31:15.294028+0800 0x271      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW return success
2020-03-20 02:31:15.294554+0800 0x271      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW wakeupOn
2020-03-20 02:31:15.294560+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: load firmware done
2020-03-20 02:31:15.294565+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: Paging: paging enabled (size = 241664 bytes)
2020-03-20 02:31:15.294572+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: wakeupOn
2020-03-20 02:31:15.294581+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_start_fw ucode_type=1
2020-03-20 02:31:15.294585+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_apm_init
2020-03-20 02:31:15.294623+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: L1 Disabled - LTR Enabled
2020-03-20 02:31:15.294696+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_load_firmware
2020-03-20 02:31:15.294708+0800 0x22599    Default     0x0                  0      0    kernel: (itlwm) itlwm: tsleep_nsec iwmfw

sixian, 2020-03-20:
try unplug your usb wifi adapter and reboot.

htmambo, 2020-03-20:
拔掉USB Wifi并且不加载驱动后重启，还是一样的报错。
Unplugging the USB Wifi and restarting without loading the driver still reports the same error.
```
kernel: (itlwm) itlwm: Asked to allocate 2048 bytes with align=256  
kernel: (itlwm) itlwm: Got allocated at paddr=0x36677000, vaddr=0x3a8b7000  
kernel: (itlwm) itlwm: Re-aligned DMA memory to paddr=0x36677000, vaddr=0x3a8b7000  
kernel: (itlwm) itlwm: config ieee80211  
kernel: (itlwm) ieee80211_ifattach  
kernel: (itlwm) itlwm: ieee80211_channel_init  
kernel: (itlwm) itlwm: ieee80211_setmode:  
kernel: (itlwm) itlwm: no channels found for mode 0  
kernel: (itlwm) itlwm: ieee80211_set_link_state  
kernel: (itlwm) itlwm: ieee80211_media_init  
kernel: (itlwm) itlwm: attach succeed.  
kernel: (itlwm) itlwm: iwm_init  
kernel: (itlwm) itlwm: iwm_init_hw  
kernel: (itlwm) itlwm: iwm_preinit  
kernel: (itlwm) itlwm: iwm_prepare_card_hw  
kernel: (itlwm) itlwm: iwm_start_hw  
kernel: (itlwm) itlwm: iwm_prepare_card_hw  
kernel: (itlwm) itlwm: iwm_apm_init  
kernel: (itlwm) itlwm: itlwm: L1 Disabled - LTR Enabled  
kernel: (itlwm) itlwm: iwm_init_msix_hw  
kernel: (itlwm) itlwm: iwm_conf_msix_hw  
kernel: (itlwm) itlwm: iwm_run_init_mvm_ucode justnvm=1  
kernel: (itlwm) itlwm: iwm_load_ucode_wait_alive ucode_type=1  
2020-03-20 15:02:44.971377+0800 0x289      Error       0x0                  61     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67050) allowed: <OSKext 0x6000015e9080 [0x7fff92b368a0]> { URL = "file:///private/tmp/itlwm.kext/", ID = "com.zxystd.itlwm" }  
2020-03-20 15:02:45.331859+0800 0x289      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW callback ret=0x00000000 length=2678284  
2020-03-20 15:02:45.331864+0800 0x289      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW return success  
kernel: (itlwm) itlwm: load firmware done  
kernel: (itlwm) itlwm: itlwm: Paging: paging enabled (size = 241664 bytes)  
kernel: (itlwm) itlwm: wakeupOn  
kernel: (itlwm) itlwm: iwm_start_fw ucode_type=1  
kernel: (itlwm) itlwm: iwm_apm_init  
kernel: (itlwm) itlwm: itlwm: L1 Disabled - LTR Enabled  
kernel: (itlwm) itlwm: iwm_load_firmware  
kernel: (itlwm) itlwm: tsleep_nsec iwmfw
```

zxystd, 2020-03-20:
是谁告诉你跟9560的固件一样的...

是这样的,9系也分两代,目前只支持一代设备,二代设备暂时没有时间去支持,目前需要先解决RX TX的大问题, 像ax200,ax201都属于二代设备,有些9560也是属于二代设备,这取决于你的pci id,跟它叫什么名字没有什么太大关系. 我查了一下表

[image]

02f0属于二代设备.

htmambo, 2020-03-20:
好吧，我其实是看着
https://www.intel.com/content/www/us/en/support/articles/000005511/network-and-i-o/wireless-networking.html
猜的，看它好像下载的固件是相同的。
[image]
那就只能继续等等待老大更新了。

closed 2020-03-20 by htmambo