OpenIntelWireless/itlwm #11: ac9560

yiseni, 2020-03-20:
你好，我想问一下，在远景论坛我看到已经支持9系的wifi,但为什么在源码中并没有9系的驱动？

williambj1, 2020-03-20:
https://github.com/zxystd/itlwm/tree/feature-openbsd-newest

yiseni, 2020-03-20:
谢谢

quan3c10, 2020-03-20:
> 谢谢

Have you try the newest? Does it work with 9560?

yiseni, 2020-03-20:
ac9560 is recognized and driven normally in the pci directory, but it cannot be connected to the network, and the startup and shutdown speed is slow, and you need to wait.

quan3c10, 2020-03-20:
Sure, I am using the IntelBluetooth kext from @zxystd and believe he will make the Wifi works soon.

mn19941219, 2020-03-20:
> ac9560 is recognized and driven normally in the pci directory, but it cannot be connected to the network, and the startup and shutdown speed is slow, and you need to wait.

Try to install kext to L/E not clover/Kext/Other

yiseni, 2020-03-20:
I will try,thanks.

MarshalX, 2020-03-20:
> > ac9560 is recognized and driven normally in the pci directory, but it cannot be connected to the network, and the startup and shutdown speed is slow, and you need to wait.
> 
> Try to install kext to L/E not clover/Kext/Other

I think it`s very bad idea) Use kextload only

closed 2020-03-21 by yiseni

quan3c10, 2020-03-21:
> I will try,thanks.

Does this work? As you closed this issue.

williambj1, 2020-03-21:
It works for me, 9560ac 8086;A370.

You do need some skills to load the kext properly.
Not quite stable, don't expect too much.

quan3c10, 2020-03-24:
> It works for me, 9560ac 8086;A370.
> 
> You do need some skills to load the kext properly.
> Not quite stable, don't expect too much.

It better to hear that. So it is just the time to make it fully works, I will wait till the stable version is released.