OpenIntelWireless/itlwm #13: how can I install it?

abdullah-abunada, 2020-03-21:
Hello,

thank you for the amazing effort, but can you clarify how can we install it, please?

Best,

vladdenisov, 2020-03-21:
Hello! 
You can try instructions [here](https://github.com/johnnync13/Xiaomi-Mi-Air/blob/master/WIFI/README.md).
If you have AC 8260 download kext from [here](https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/330#issuecomment-600749608).

williambj1, 2020-03-21:
The kext is currently at a very early stage, extremely unstable and can't be considered usable, so the developer doesn't expect anyone who doesn't have debugging skills to mess up with their systems. Of course, you have the right to give it a try.

abdullah-abunada, 2020-03-21:
Hi,
I have this card, Intel Wireless 3168

williambj1, 2020-03-21:
Build, the kext yourself, don't ask how, a simple google search will tell you.
Don't expect too much from this kext at this stage.

abdullah-abunada, 2020-03-21:
I downloaded itlwm.kext and run the following commands:

` sudo chmod -R 755 itlwm.kext`
`sudo kextload -v itlwm.kext `
and the results was
`itlwm.kext loaded successfully (or already loaded).`

But unfortunately its not working and no wifi driver

abdullah-abunada, 2020-03-21:
After reading, this step was missed:
**For itlwm u need to create open (without password) hotspot with SSID "Redmi"**

closed 2020-03-21 by abdullah-abunada