OpenIntelWireless/itlwm #14: Intel AC 9560

trankimlonghai, 2020-03-22:
Hi, I wonder if Intel AC 9560 is supported ?

yiseni, 2020-03-22:
pciid:A370 is ok.

trankimlonghai, 2020-03-22:
> pciid:A370 is ok.

Oh whoa, okay, that's good new. So... Can you give me the link to the guide how to build/install the kext ?

yiseni, 2020-03-22:
Please see closed issues .

wwwsky, 2020-03-22:
英特尔AC 9560  log

2020-03-22 22:17:10.130000+0800 0x22dae    Default     0x0                  1254   0    sudo:     ding : TTY=ttys000 ; PWD=/Users/ding/Desktop ; USER=root ; COMMAND=/usr/sbin/chown -R root:wheel /tmp/itlwm.kext
2020-03-22 22:17:13.860089+0800 0x23047    Default     0x0                  1256   0    sudo:     ding : TTY=ttys000 ; PWD=/Users/ding/Desktop ; USER=root ; COMMAND=/usr/bin/kextutil /tmp/itlwm.kext
2020-03-22 22:17:14.058922+0800 0x2304d    Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-03-22 22:17:14.061529+0800 0x23085    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.061550+0800 0x23085    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.061552+0800 0x23085    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.061570+0800 0x23085    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.061986+0800 0x23090    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062001+0800 0x23090    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062003+0800 0x23090    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062010+0800 0x23090    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062061+0800 0x23087    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062075+0800 0x23087    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062077+0800 0x23087    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062084+0800 0x23087    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062123+0800 0x2308c    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062139+0800 0x2308c    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062141+0800 0x2308c    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062151+0800 0x2308c    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062174+0800 0x23096    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062189+0800 0x23096    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062191+0800 0x23096    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062205+0800 0x23096    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062208+0800 0x23083    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062226+0800 0x23083    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062228+0800 0x23083    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062233+0800 0x23083    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062325+0800 0x23086    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062338+0800 0x23086    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062340+0800 0x23086    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062345+0800 0x23086    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062404+0800 0x23088    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062416+0800 0x23088    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062418+0800 0x23088    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062428+0800 0x23088    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.062590+0800 0x2308a    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.062602+0800 0x2308a    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.062604+0800 0x2308a    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.062618+0800 0x2308a    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.185282+0800 0x230b2    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.185291+0800 0x230b2    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.185292+0800 0x230b2    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.185300+0800 0x230b2    Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-03-22 22:17:14.212894+0800 0x230b8    Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-03-22 22:17:14.212904+0800 0x230b8    Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-03-22 22:17:14.212906+0800 0x230b8    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-03-22 22:17:14.212921+0800 0x230b8    Default     0x0                  0      0    kernel: (itlwm) itlwm: free

trankimlonghai, 2020-03-22:
[image]
How can I resolve this ?

zxystd, 2020-03-22:
> [image]
> How can I resolve this ?

try "sudo kextutil -v itlwm.kext" command and see the error log.

htmambo, 2020-03-22:
> 英特尔AC 9560 log
>
> 2020-03-22 22:17:10.130000+0800 0x22dae Default 0x0 1254 0 sudo: ding : TTY=ttys000 ; PWD=/Users/ding/Desktop ; USER=root ; COMMAND=/usr/sbin/chown -R root:wheel /tmp/itlwm.kext [...]

看你这个，应该也是二代设备，固件不支持，还得等作者继续更新。

trankimlonghai, 2020-03-22:
> > [image]
> > How can I resolve this ?
> 
> try "sudo kextutil -v itlwm.kext" command and see the error log.

[image]
Here's the log ^^

williambj1, 2020-03-22:
It's really not recommended to try this kext if you don't have any debugging or hackintoshing skills.

trankimlonghai, 2020-03-22:
> It's really not recommended to try this kext if you don't have any debugging or hackintoshing skills.

But my friends they use the kext normally... I just curious and dont know how to use it correctly so that means I want to learn it regardless about the problem of where the skills level are

MarshalX, 2020-03-22:
> > It's really not recommended to try this kext if you don't have any debugging or hackintoshing skills.
> 
> But my friends they use the kext normally... I just curious and dont know how to use it correctly so that means I want to learn it regardless about the problem of where the skills level are

Why don't you ask your friends?

trankimlonghai, 2020-03-22:
> Why don't you ask your friends?

If you think you can help, then help me out. If not, please don't ask me like that.
Ask my friends or the hackintosh community or whoever I want, that's why I can gain more my knowledge about the hackintosh from foreigners. Not only in my country, my country they don't care if they will help or not. They just care about the service they can do to help install the hackintosh on customer's machines.

williambj1, 2020-03-22:
Do you think such a brilliant coder would let people not using his kext by not providing releases if the kext is fully usable? The developer doesn't want people without skills to reach the binary files!

This kext is currently extremely unstable, panics very often, and you'll only have about 1Mbps download and 0.01Mbps upload. Long way to go!

trankimlonghai, 2020-03-22:
Okay, thanks a lot for your care. I do understand. I am not struggle and insane that I shout at people like you said. I am in normal and good mood. You just overthinking. I am just asking for help, I am not a boss. Hope you understand. If I did or said something not right to you and all devs. I am very sorry. I don't mean that.

williambj1, 2020-03-22:
GitHub Issues is a place to POST BUGS, not for requesting help, especially from a developer not even responsible for your personal lack of skills.

I've been testing with 9560ac with zxystd for quite a long time. He has a solid understanding of what is going on. We speak the same language which makes the progress much easier. I understand you passion for this great project, but at this stage, it's better to do some research first and stay tune.

zxystd, 2020-03-22:
Stop talking about "How to use" or "Hackintosh primary skills" topics, these can help me anything and I don't have the ability to answer these meaningless questions. I am going to close this issue. Thanks.

closed 2020-03-22 by zxystd

startpenghubingzhou, 2020-03-22:
> [image]
> How can I resolve this ?

 Hey thx for your starring. It's good to see your enthusiasm in this project and hackintosh. But as we all know，it's necessary to have some basic skills and  acknowledgement in hackintosh and kext ，aka IOKit kernel extensions debuging and testing，so that you can get understand all the things about helping development. So if you can't ensure you have these things，I  think it's better to wait for developer' s implement， isn't it？