OpenIntelWireless/itlwm #18: 8265AC，连接成功后第一次打开浏览器能上网，之后浏览器卡死

okexi, 2020-03-24:
机型小米Pro，系统15.3，表现为浏览器打开第一个标签可以正常加载，之后的标签不能加载，然后浏览器卡死，强制结束后再不能打开浏览器，打开直接转圈（两个浏览器edge和safari均是），今天想复现问题但是一加载就死机了。
浏览器卡死的那次日志如下 
[log.zip](https://github.com/zxystd/itlwm/files/4372681/log.zip)
> 日志有点多，稍微编辑了一下，省略了很多 Map new address 在182行

deniro98, 2020-03-25:
Similarly, the maximum speed does not exceed 9 Mbps, and  it is very unstable

zb2947244682, 2020-03-31:
膜拜大佬～～

zxystd, 2020-04-07:
Yes, this problem is really exists, and I am trying to fix it.

itdawn, 2020-04-09:
I am also this network card, after using the latest driver, did not appear this kind of situation, but the network speed is very slow.    @zxystd 
[image]

zxystd, 2020-05-29:
issue fix in fbd5b23, thanks.

closed 2020-05-29 by zxystd