OpenIntelWireless/itlwm #27: AC 9462 Not detecting Wifi

Arrnav, 2020-04-07:
Wifi card : Intel AC 9462
Laptop : Dell G3 3579 i5
MacOS 10.15.4 with Opencore

Changed SSID details in source -
SSID Name : "Private Network"
SSID Password : ""

Built latest itlwm kext and installed to /L/E/ with Hackintool (Injecting the kext with OpenCore caused a /L/E/ realpath error with "log show --last boot | grep itlwm" in terminal) and rebooted.

While Laptop was rebooting, turned on Personal Hotspot on my Galaxy S10 with settings -

Name : "Private Network"
Security : Open
Password : None

Once laptop booted, opened System Preferences > Network and the window is empty with no network connections shown at all, then the dialog box hangs for 5 minutes and finally shows two Ethernet connections and none of them are connected.

Tried turning off and on mobile hotspot but still no ethernet shows as connected on laptop.

Obviously internet is not working.

Checked whether kext is loaded properly by using "sudo kextstat | grep itlwm" in terminal and got following output -

`arnavmazumder@Arnavs-MacBook-Pro ~ % sudo kextstat | grep itlwm       
Password:
  125    1 0xffffff7f83075000 0x80000    0x80000    com.zxystd.itlwm (1) 2C4B068D-3F35-3C5C-82CC-178D5E52E1CA <18 13 6 5 3 1>
arnavmazumder@Arnavs-MacBook-Pro ~ % 
`
Then to check complete log in terminal "log show --last boot | grep itlwm" and output was -

`Last login: Tue Apr  7 23:28:10 on ttys000
arnavmazumder@Arnavs-MacBook-Pro ~ % log show --last boot | grep itlwm
2020-04-07 22:57:15.748518+0530 0x4ec      Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-04-07 22:57:15.748626+0530 0x4ec      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.748652+0530 0x4ec      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.748654+0530 0x4ec      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.748688+0530 0x4ec      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:15.748724+0530 0x4e6      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.748725+0530 0x4e7      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.748731+0530 0x4e6      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.748748+0530 0x4e6      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.748755+0530 0x4e6      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:15.748758+0530 0x4e7      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.748760+0530 0x4e7      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.748775+0530 0x4e7      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:15.749315+0530 0x4e3      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.749326+0530 0x4e3      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.749327+0530 0x4e3      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.749335+0530 0x4e3      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:15.749369+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.749377+0530 0x4e1      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:15.749386+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.749400+0530 0x4e1      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:15.749402+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.749403+0530 0x4e1      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:15.749411+0530 0x4e1      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:15.749443+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: start
2020-04-07 22:57:15.749590+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_attach
2020-04-07 22:57:15.749781+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: , (null)
2020-04-07 22:57:15.749815+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-04-07 22:57:15.750652+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: alloc contig
2020-04-07 22:57:15.750653+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Asked to allocate 327680 bytes with align=16
2020-04-07 22:57:15.750737+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Got allocated at paddr=0xa666000, vaddr=0x7e8f5000
2020-04-07 22:57:15.750739+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Re-aligned DMA memory to paddr=0xa666000, vaddr=0x7e8f5000
2020-04-07 22:57:15.750742+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Asked to allocate 4096 bytes with align=4096
2020-04-07 22:57:15.750746+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Got allocated at paddr=0xa249000, vaddr=0x7b1cd000
2020-04-07 22:57:15.750748+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Re-aligned DMA memory to paddr=0xa249000, vaddr=0x7b1cd000
2020-04-07 22:57:15.750751+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Asked to allocate 4096 bytes with align=4096
2020-04-07 22:57:15.750754+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: Got allocated at paddr=0xa248000, vaddr=0x7b1ce000
[... 699 lines cut ...]
2020-04-07 22:57:15.796186+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_conf_msix_hw
2020-04-07 22:57:15.796223+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_run_init_mvm_ucode justnvm=1
2020-04-07 22:57:15.796225+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_load_ucode_wait_alive ucode_type=1
2020-04-07 22:57:16.164467+0530 0x4ca      Default     0x0                  0      0    kernel: (itlwm) itlwm: init
2020-04-07 22:57:16.164480+0530 0x4ca      Default     0x0                  0      0    kernel: (itlwm) itlwm: probe
2020-04-07 22:57:16.164488+0530 0x4ca      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-07 22:57:16.164494+0530 0x4ca      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-07 22:57:16.407045+0530 0x249      Error       0x0                  66     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67050) allowed: <OSKext 0x60000113f2a0 [0x7fff8ac588c0]> { URL = "file:///Library/Extensions/itlwm.kext/", ID = "com.zxystd.itlwm" }
2020-04-07 22:57:16.648493+0530 0x249      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW callback ret=0x00000000 length=2678284
2020-04-07 22:57:16.648497+0530 0x249      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW return success
2020-04-07 22:57:16.648854+0530 0x249      Default     0x0                  0      0    kernel: (itlwm) itlwm: onLoadFW wakeupOn
2020-04-07 22:57:16.649530+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: load firmware done
2020-04-07 22:57:16.649536+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: Paging: paging enabled (size = 241664 bytes)
2020-04-07 22:57:16.649540+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: wakeupOn
2020-04-07 22:57:16.649544+0530 0x4e4      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_start_fw ucode_type=1
[... cut ...]
2020-04-07 22:58:58.390471+0530 0xc66      Error       0x0                  559    0    kextcache: [com.apple.kext:kextlog] Kext with invalid signature (-67050) allowed: <OSKext 0x7fda385062d0 [0x7fff8ac588c0]> { URL = "file:///Library/Extensions/itlwm.kext/", ID = "com.zxystd.itlwm" }
2020-04-07 22:59:03.579500+0530 0xc66      Error       0x0                  559    0    kextcache: [com.apple.kext:kextlog] Invalid signature -67050 for kext <OSKext 0x7fda385062d0 [0x7fff8ac588c0]> { URL = "file:///Library/Extensions/itlwm.kext/", ID = "com.zxystd.itlwm" }
arnavmazumder@Arnavs-MacBook-Pro ~ % 
`

Where am I going wrong?

lgariv, 2020-04-08:
you need to set your ssid and password prior to compiling / set you ssid to "ssdt" (without quotes) and password to "zxyssdt112233" (without quotes).
then, load the kext and give it a minute to do it's magic

williambj1, 2020-04-08:
You can't disable encryption by simply setting the password into an empty string.

Arrnav, 2020-04-08:
> you need to set your ssid **and password** prior to compiling / set you ssid to "ssdt" (without quotes) and password to "zxyssdt112233" (without quotes).
> then, load the kext and give it a minute to do it's magic

But setting the password to anything automatically changes the security level to WPA2 PSK, which as far as I read, is not supported yet right?

Anyhow, I've tried compiling with setting the ssid name and password as you mentioned, loaded the kext, gave it 10 minutes on boot and nothing. Again the network doesn't connect. 

SSID Name : ssdt
Password: zxyssdt112233
Security : WPA2 PSK (Encryption cannot be set to any other setting when ANY password is set, on two different devices I tried)

Logs for kexts are same as in original post.

zxystd, 2020-04-08:
device product id?

zxystd, 2020-04-08:
yesterday i have merge a pr that add some pids which are not tested.

Arrnav, 2020-04-08:
[image]

Well I wasn't sure which ID you needed exactly, so here's a screenshot of the Hardware ID in windows device manager, I can provide anything else if needed.

If the above is wrong, where do I exactly find this ID? @zxystd

zxystd, 2020-04-08:
Yes, 0xA370 is your device productid, this id is same as AC9560... but My friend has test it and it works well, sometimes stuck when loading firmware, maybe you can try about 3~5 times and see whether it would work? I will mark it as a bug and will fix it(I think it is a irq issue).

Arrnav, 2020-04-08:
> yesterday i have merge a pr that add some pids which are not tested.

I just tried your latest code and rebuilt and installed the kext and still does not work.

Arrnav, 2020-04-08:
1. How does the kext scan for wifi network? Does it keep scanning after boot? For example, if I disable hotspot and enable after 30 mins, will the laptop still try to connect again?

2. I have working bluetooth using IntelBluetoothFirmware.kext and IntelBluetoothInjection.kext for the same card. Can this cause any interference in the working of itlwm?

3. Just to confirm again, my mobile hotspot gets automatically set to WPA2 PSK if setting password as per the kext. Will this create an issue?

zxystd, 2020-04-08:
1.If the card is init done, it will enter a scheduling scan work, scan for every channel until find the target ssid and connect it.
2.Not too much relation between these things.
3.WPA2 is fully support now, also including WPA-PSK WiFi.

Arrnav, 2020-04-08:
How do I know whether the card init is done properly and it's scanning? @zxystd

lgariv, 2020-04-08:
> How do I know whether the card init is done properly and it's scanning? @zxystd

By reading logs

zxystd, 2020-04-08:
log show --last boot|grep scan_result

and you will see the surrounding wifis if the scan works.

zxystd, 2020-04-08:
also, if you can see a network interface is created by driver, that means everything init done.

Arrnav, 2020-04-10:
I'll check the scanning log and let you know.

Is there any way to install this kext just on the OpenCore EFI folder? (I want to preserve native Mac /L/E/ by injecting all third party kexts in OpenCore)

Everytime I install itlwm.kext to OpenCore/EFI/ instead of /L/E/, I get a "realpath not found in /L/E/" error in the results of "log show --last boot | grep itlwm" in terminal, maybe that interferes with kext loading too.

When I install the kext in /L/E, the error is gone in next log.

Regardless of where it's installed now, still doesn't detect hotpost.

zxystd, 2020-04-10:
Of course yes, just need some minor of change to do because of the OSKextRequestResource api can not work in bootloader more(maybe).

tonyleelyy, 2020-04-12:
> also, if you can see a network interface is created by driver, that means everything init done.

大佬您好，上面这位@Arrnav 国外老哥使用的是我的OC项目（之前跟他交流过），跟我是同一机型，使用AC9462。

而我在我的机器上面测试，也是跟他一样的问题，驱动已加载（在关于本机-系统报告-功能扩展里已经看到itlwm），就是在设置的网络里，没有除了有线以太网和蓝牙PAN之外的东西，无法上网。使用您上面提供的scan_result代码运行之后没有任何信息提供。

而我注意到在Hackintool里，PCIe一栏下Wireless-AC 9462显示的是已禁用，系统报告的PCI一栏也没有这一网卡的信息，**我就在想是否是我的OC配置有问题？**

我也尝试过在Config.plist注入网卡的pci信息，系统报告里的pci是有这一网卡了，但是仍然无法使用您的驱动，hackintool里PCIe下网卡也是显示已禁用。

小弟不太懂，提供我的EFI给大佬参考：https://github.com/tonyleelyy/OpenCore-Hackintosh-Dell-G3-3579/releases/tag/v1.2

log文件也一并提供：[test.log](https://github.com/zxystd/itlwm/files/4465848/test.log)

希望对大佬的研究有帮助。祝成功！

tonyleelyy, 2020-04-12:
大佬不好意思，我已经确定是我的EFI存在问题，现已修复。

衷心感谢您的付出和努力！

Arrnav, 2020-04-12:
> 大佬不好意思，我已经确定是我的EFI存在问题，现已修复。
> 
> 衷心感谢您的付出和努力！

Can you tell me what the problem is brother?

Did you manage to fix the wifi?

Arrnav, 2020-04-12:
> 大佬不好意思，我已经确定是我的EFI存在问题，现已修复。
> 
> 衷心感谢您的付出和努力！

Did you upload the new fixed EFI to your OpenCore GitHub Release?

tonyleelyy, 2020-04-12:
> > 大佬不好意思，我已经确定是我的EFI存在问题，现已修复。
> > 衷心感谢您的付出和努力！
> 
> Did you upload the new fixed EFI to your OpenCore GitHub Release?

Yes, the newest version can use this kext successfully!

I think it is something about the injection of devices in my older ACPI files (the older version is build by my friend CerteKim)

closed 2020-05-29 by zxystd