OpenIntelWireless/itlwm #29: After downloading about 5MB of data, the system crashes.

zdypk1994, 2020-04-09:
After downloading about 5MB of data, the system crashes.
This problem can be reproduced 100%.
Wireless device:AC9560
Laptop:HUAWEI Matebook14 2019
System:Mac OS 10.15.3
Thank you for your dedication to the Hackintosh community and hopefully this bug will be fixed soon.

zxystd, 2020-04-09:
What about "system crashes"? freeze or reboot? any logs? card device product id? these info can help me locate the cause, thanks.

pytason, 2020-04-09:
I have the same Card 
Wireless device: AC9560
And can confirm I have the same issue.
here is my partial log

[log.txt](https://github.com/zxystd/itlwm/files/4455697/log.txt)

zxystd, 2020-04-12:
> @zxystd
> I have the same Card
> Wireless device: AC9560
> And can confirm I have the same issue.
> here is my partial log
> 
> [log.txt](https://github.com/zxystd/itlwm/files/4455697/log.txt)

> After downloading about 5MB of data, the system crashes.
> This problem can be reproduced 100%.
> Wireless device:AC9560
> Laptop:HUAWEI Matebook14 2019
> System:Mac OS 10.15.3
> Thank you for your dedication to the Hackintosh community and hopefully this bug will be fixed soon.

Seems firmware had been crash according to the log, it is a known issue and I am trying to fix it, After that, the driver will be working normally.

pytason, 2020-04-22:
@zxystd 
Wireless device: AC9560
8086:0034

Thank you, now download speed is much better. 
Now it crashes after about 1 hour of usage, I was able to download about 500MB.

[log.txt](https://github.com/zxystd/itlwm/files/4513845/log.txt)

zxystd, 2020-05-29:
upload issue fix in fbd5b23, thanks.

closed 2020-05-29 by zxystd