OpenIntelWireless/itlwm #31: Network speed is slow

itdawn, 2020-04-10:
I successfully drove the 8265AC with the latest kext, but the network speed was slow.
[image]

epictgames, 2020-04-10:
how did you build the kext?

zxystd, 2020-04-22:
Now the network speed is much better, you can try. Upload speed is still slow.

sixian, 2020-04-22:
> Now the network speed is much better, you can try. Upload speed is still slow.

9560ac 用最新的build不能连接网络了,上一个版本可以的
[log.txt](https://github.com/zxystd/itlwm/files/4513714/log.txt)

williambj1, 2020-04-22:
@sixian 

1. 打开系统的偏好设置，删掉网络里所有的端口，右下角应用
1. 终端输入 ```open /Library/Preferences/SystemConfiguration```
1. 删掉
   - **`com.apple.airport.preferences.plist`**
   - **`preferences.plist`**
   - **`NetworkInterfaces.plist`**
   - **`com.apple.network.identification.plist`**
   - **`com.apple.wifi.message-tracer.plist`**
   - 以及目录下所有包含 `enX` 的文件，(可以用 grep 判断)
   - (如果更换过机型，会有个以机型开头的文件，也要删掉)
1. 关机后**彻底断电**，再开机
1. 保险起见，提前备份，这些文件都是自动创建的，本来没有特殊设置的话不会出问题。

zxystd, 2020-04-22:
> > Now the network speed is much better, you can try. Upload speed is still slow.
> 
> 9560ac 用最新的build不能连接网络了,上一个版本可以的
> [log.txt](https://github.com/zxystd/itlwm/files/4513714/log.txt)

是不是WPA2-PSK的网络？WPA1的一直都有点问题，你可以先用WPA2的网络测试

sixian, 2020-04-22:
> @sixian
> 
> 1. 打开系统的偏好设置，删掉网络里所有的端口，右下角应用
> 2. 终端输入 `open /Library/Preferences/SystemConfiguration`
> 3. 删掉
>    
>    * **`com.apple.airport.preferences.plist`**
>    * **`preferences.plist`**
>    * **`NetworkInterfaces.plist`**
>    * **`com.apple.network.identification.plist`**
>    * **`com.apple.wifi.message-tracer.plist`**
>    * 以及目录下所有包含 `enX` 的文件，(可以用 grep 判断)
>    * (如果更换过机型，会有个以机型开头的文件，也要删掉)
> 4. 关机后**彻底断电**，再开机
> 5. 保险起见，提前备份，这些文件都是自动创建的，本来没有特殊设置的话不会出问题。

谢谢! 现在可以连接了,可能是之前用过其他的驱动导致的

zxystd, 2020-05-29:
Now it fix in fbd5b23, thanks.

closed 2020-05-29 by zxystd

JasonOoi20, 2020-06-13:
Thanks. This really works on my case on my AC 9560.

mehmetyukcell, 2020-06-13:
> Thanks. This really works on my case on my AC 9560. 

How install kext? Or Which kext?

JasonOoi20, 2020-06-14:
> > Thanks. This really works on my case on my AC 9560. 
> 
> How install kext? Or Which kext?

Just sideload the itlwm.kext using console or put it inside the clover kext/other/ folder.

Cxp1220, 2020-08-11:
大佬，我现在用的itlwm.kext，网速比较慢，插上网线速度也不正常了，我想暂时停用itlwm，然后就把它关了，但是关了之后有线也连不上网了，怎么解决

Cxp1220, 2020-08-16:
为啥不回复我

williambj1, 2020-08-16:
Off topic