OpenIntelWireless/itlwm #33: Can't find anything after load the kext (Intel 9560)

dlhxr, 2020-04-13:
I compiled the latest kext and loaded it, but didn't manage to make it  work for my intel 9560.
Here's the log.
[log.log](https://github.com/zxystd/itlwm/files/4472260/log.log)

zxystd, 2020-04-14:
重启再加载一次试试

dlhxr, 2020-04-14:
结果也是一样，又是到iwm_load_firmware就没有了。。。
[log2.log](https://github.com/zxystd/itlwm/files/4472845/log2.log)

zxystd, 2020-04-14:
product id是多少？

dlhxr, 2020-04-14:
A370，我看源码里面有这个id啊。。。

dlhxr, 2020-04-14:
IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/CNVW@14,3
PciRoot(0x0)/Pci(0x14,0x3)
倒是我同时用HoRNDIS.kext，连着手机上网，也是生成了一个ethernet，冲突？

zxystd, 2020-04-14:
这个id是支持的，我朋友的也是这个id，他可以用，不过也是有时候跟你一样会卡在这，重启几次可以。嗯，你也可以试试把HORNDIS创建的网络接口删了再试试。

dlhxr, 2020-04-14:
试过删掉HORNDIS的接口，而且前面测试的时候都没插usb手机。我再多试试看吧，有进展再更新，谢谢啦！

yxh139270, 2020-04-20:
我的也是9560，开始也是加载了没反应，多试几次就好了。现在就是网速非常慢。能正常驱动。

dlhxr, 2020-04-22:
和https://github.com/zxystd/itlwm/issues/39一样，BIOS里面开关一次就能驱动…

closed 2020-04-22 by dlhxr