OpenIntelWireless/itlwm #36: Intel AX200 Support

Malumen, 2020-04-15:
Would it be possible to get Intel WiFi chip AX200 working?
https://ark.intel.com/content/www/us/en/ark/products/189347/intel-wi-fi-6-ax200.html
Many systems (desktops, laptops) have M.2 WiFi slots, and this product is very cheap!

可以使用Intel AX200 吗？
很多电脑有M.2。 Intel AX200 设备价格便宜。

Stevemoretz, 2020-04-19:
If you want something cheap why not get a USB dongle wifi? The whole point of this project is to get the built in wifis work if you want to replace your wifi card why not replace it with a supported non Intel card.Cheap wifi cards aren't very high quality and I suggest you stay away from it anyway.

aluveitie, 2020-04-19:
I think most if not even all X570 Motherboards with Wifi 6/BT 5 I've looked at had the Intel AX200.

zxystd, 2020-04-19:
Ax200 is supporting now..., now it can scan, auth, but remains some problems...

aluveitie, 2020-04-19:
@zxystd wow, that's great to hear. Thanks for your continuous hard work!

aluveitie, 2020-04-19:
@zxystd I'v compiled the latest version and added the kext to my OpenCore testing boot stick but it does not seem to work just yet:
`log show --last boot | grep itlwm`:
```
2020-04-19 15:22:35.587948+0200 0x136      Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-04-19 15:22:35.588015+0200 0x136      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-19 15:22:35.588025+0200 0x136      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-19 15:22:35.591221+0200 0x16b      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-19 15:22:35.591239+0200 0x16b      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-19 15:22:35.591439+0200 0x16f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-19 15:22:35.591480+0200 0x16f      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-19 15:22:43.791254+0200 0x5bc      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-19 15:22:43.791259+0200 0x5bc      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
```

zxystd, 2020-04-19:
I have not push the newest code as it's problems...

aluveitie, 2020-04-19:
@zxystd thanks for the update, I'll watch out and re-test once you've pushed it 👍

zxystd, 2020-04-19:
It is a separate kext which call itlwmx, different from itlwm, it also support gen2 and future gen3 device.

nujabse, 2020-04-19:
Is intel AX 201 in the supporting card list?

zxystd, 2020-04-19:
Yes in theory, but I don't have ax201 device for test. so just wait for the ax200 support.

yaozaiyu, 2020-04-24:
@zxystd 现在能测试ax200了吗，我发现你推送了提交支持了ax200

zxystd, 2020-04-24:
No, code is done, but not fully debug.

Floris497, 2020-05-09:
I just bought a AX200 card to test some of the progress here. Looks like authentication does not work correctly yet. last part of my log:

```
2020-05-09 22:32:24.458111+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: tsleep_nsec iwxcmd
2020-05-09 22:32:24.458267+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: Interrupt!!!
2020-05-09 22:32:24.458270+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_notif_intr
2020-05-09 22:32:24.458272+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_rx_pkt code=0x29
2020-05-09 22:32:24.458274+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: wakeupOn
2020-05-09 22:32:24.458277+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_cmd_done: command 0x29 done
2020-05-09 22:32:24.458296+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: Interrupt!!!
2020-05-09 22:32:24.458298+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_notif_intr
2020-05-09 22:32:24.458300+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_rx_pkt code=0x2a
2020-05-09 22:32:24.458421+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : SCAN -> AUTH
2020-05-09 22:32:24.458424+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state nstate=2, old_state=2
2020-05-09 22:32:24.458431+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : sending auth to --:--:--:--:--:-- on channel 1 mode 11g
2020-05-09 22:32:24.458440+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_start 6840
2020-05-09 22:32:24.458442+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_start 6851
2020-05-09 22:32:24.458444+0200 0x18cb     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_tx
2020-05-09 22:32:24.463341+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: Interrupt!!!
2020-05-09 22:32:24.463347+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_notif_intr
2020-05-09 22:32:24.480639+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: Interrupt!!!
2020-05-09 22:32:24.480647+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_notif_intr
2020-05-09 22:32:24.480652+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_rx_pkt code=0x1c
2020-05-09 22:32:24.480661+0200 0x70       Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_start 6840
2020-05-09 22:38:46.830252+0200 0x2f0      Default     0x0                  0      0    kernel: (itlwmx) itlwm: outputPacket
```

As i understood i do not need the `itl80211.kext` correct? because it does not load for me anyway..

And status: **Cable Unplugged** `Either the cable for Thunderbolt Ethernet Slot  2 is not plugged in or the device at the other end is not responding.`

Running this on a Thunderbolt to PCIe box

I will try to invest some time in reading the source code and if i can get enough familiar with it contribute to this project.

M69W, 2020-05-21:
[OpenBSD 6.7](https://www.openbsd.org/67.html)
Improved hardware support, including:
...
Introduced iwx(4), a driver for Intel AX200 WiFi devices.

RGarrido03, 2020-05-22:
I have an Intel AX200 card, I can test whatever you want!
There’s an error in itlwmx.kext tho, I will update this post with more details when possible

jianghaizhi, 2020-05-30:
Could you also add the support of Killer ax1650? This card has the same hardware with AX200

RGarrido03, 2020-05-30:
Built new itlwmx kext, loaded successfully and my AX200 appears as Ethernet 2 on Network preferences. Also, on Hackintool it shows Intel AX200 as well as Realtek Ethernet card. However, it doesn’t connect (says “Not connected”)
Please tell me what logs do you need, I can provide :)
However, thank you so much for all your effort!

zxystd, 2020-05-31:
Yes, it it init done, and can scan, but exits problems:
1. when output frame data, the firmware crash;
2. scan crash randomly;
3. didn't support active scan;
4. didn't support backgroud scan.

I had try install openbsd and test, *Confirm that these problems exists in openbsd*, it is fully unstable. so, just wait for fix, I am going to do other things such as itlwm speed improvement and client development.

zxystd, 2020-05-31:
And maybe some devices can work. you can send me the log and I can see what happened.

log show --last boot | grep itlwm >itlwmx.log

RGarrido03, 2020-06-01:
> And maybe some devices can work. you can send me the log and I can see what happened.
> 
> log show --last boot | grep itlwm >itlwmx.log

Hi! Here it is. Thank you so much!!
PS: Using AX200 and latest commit's itlwmx kext
[itlwmx.log](https://github.com/zxystd/itlwm/files/4711184/itlwmx.log)

zxystd, 2020-06-01:
Thanks for your test. According to the log, I can see the card is full init and begin scan and get scanned  result.
MEO-64C990
MEO-64C991-5G
MEO-WiFi
MEO-WiFi

but when send auto to 'MEO-64C991-5G', the firmware crash. just the same as my device.

fudaming, 2020-06-02:
Seems that can not start to scan the ssid. the device is ax200(id 0x2723),original shipment on ASRock X570 Phontom Gaming-ITX/TB3.

[image]

[itlwmx.log](https://github.com/zxystd/itlwm/files/4717127/itlwmx.log)

yaozaiyu, 2020-06-04:
能提供固件给我们测试吗?不太会打包

fudaming, 2020-06-04:
[itlwm.log](https://github.com/zxystd/itlwm/files/4729529/itlwm.log)
👆this is the log that rebuild the kext again, and bad news is when kextunload, the screen is black out, and system reboot.

the good news is the driver can find ssid all around.😂

fudaming, 2020-06-09:
always spot "fatal fireware error" in log, try to do something with the newest firmware from intel official.
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/

`
2020-06-09 19:17:06.257605+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm: driver status:
2020-06-09 19:17:06.257607+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  0: qid=0  cur=21  queued=0
2020-06-09 19:17:06.257609+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  1: qid=1  cur=0   queued=0
2020-06-09 19:17:06.257611+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  2: qid=2  cur=0   queued=0
2020-06-09 19:17:06.257613+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  3: qid=3  cur=0   queued=0
2020-06-09 19:17:06.257615+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  4: qid=4  cur=0   queued=0
2020-06-09 19:17:06.257617+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  5: qid=5  cur=0   queued=0
2020-06-09 19:17:06.257618+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  6: qid=6  cur=0   queued=0
2020-06-09 19:17:06.257620+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  7: qid=7  cur=0   queued=0
2020-06-09 19:17:06.257622+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  8: qid=8  cur=0   queued=0
2020-06-09 19:17:06.257624+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring  9: qid=9  cur=0   queued=0
2020-06-09 19:17:06.257626+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 10: qid=10 cur=0   queued=0
2020-06-09 19:17:06.257628+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 11: qid=11 cur=0   queued=0
2020-06-09 19:17:06.257631+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 12: qid=12 cur=0   queued=0
2020-06-09 19:17:06.257633+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 13: qid=13 cur=0   queued=0
2020-06-09 19:17:06.257635+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 14: qid=14 cur=0   queued=0
2020-06-09 19:17:06.257637+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 15: qid=15 cur=0   queued=0
2020-06-09 19:17:06.257639+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 16: qid=16 cur=0   queued=0
2020-06-09 19:17:06.257641+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 17: qid=17 cur=0   queued=0
2020-06-09 19:17:06.257643+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 18: qid=18 cur=0   queued=0
2020-06-09 19:17:06.257644+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 19: qid=19 cur=0   queued=0
2020-06-09 19:17:06.257647+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 20: qid=20 cur=0   queued=0
2020-06-09 19:17:06.257649+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 21: qid=21 cur=0   queued=0
2020-06-09 19:17:06.257650+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 22: qid=22 cur=0   queued=0
2020-06-09 19:17:06.257653+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 23: qid=23 cur=0   queued=0
2020-06-09 19:17:06.257655+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 24: qid=24 cur=0   queued=0
2020-06-09 19:17:06.257656+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 25: qid=25 cur=0   queued=0
2020-06-09 19:17:06.257658+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 26: qid=26 cur=0   queued=0
2020-06-09 19:17:06.257660+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 27: qid=27 cur=0   queued=0
2020-06-09 19:17:06.257662+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 28: qid=28 cur=0   queued=0
2020-06-09 19:17:06.257664+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 29: qid=29 cur=0   queued=0
2020-06-09 19:17:06.257666+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   tx ring 30: qid=30 cur=0   queued=0
2020-06-09 19:17:06.257667+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   rx ring: cur=279
2020-06-09 19:17:06.257669+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm:   802.11 state INIT
2020-06-09 19:17:06.257670+0800 0x5633     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : fatal firmware error
[... cut ...]
2020-06-09 19:17:08.443760+0800 0x562c     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-06-09 19:17:08.443788+0800 0x562c     Default     0x0                  0      0    kernel: (itlwmx) itlwm: attach to interface fail
2020-06-09 19:17:08.443791+0800 0x562c     Default     0x0                  0      0    kernel: (itlwmx) itlwm: wakeupOn
2020-06-09 19:17:08.443808+0800 0x562c     Default     0x0                  0      0    kernel:          com.zxystd.itlwmx(1.0)[954F88A2-4CED-3BA5-A105-A2C8A2103EA0]@0xffffff7f873db000->0xffffff7f8748afff
`

fudaming, 2020-06-09:
silly me. failed😆
[image]

lsvih, 2020-06-11:
Still failed on Intel 9560ac, productId 02F0. Had tried solution at #52 but it didn't work :(

[itlwmx.log](https://github.com/zxystd/itlwm/files/4767550/itlwmx.log)

fudaming, 2020-06-12:
Greate news! start to scan ssid, but can not connect to my wireless.
here is the log [itlwmx.log](https://github.com/zxystd/itlwm/files/4771459/itlwmx.log)
and network setting below.
[image]

yaozaiyu, 2020-06-12:
所以只需要放入itwmx.kext,并且改账号密码，ax200就可以联网？？ 
 版 
2020-06-12 23:39:33.223317+0800 0xd2       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.262711+0800 0xd5       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.262712+0800 0xd4       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263399+0800 0xe6       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263557+0800 0xd3       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263679+0800 0xdd       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263684+0800 0xe9       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263798+0800 0xe7       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.263828+0800 0xdc       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.280759+0800 0xe8       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.281194+0800 0xd9       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.281233+0800 0xdf       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.281314+0800 0xd7       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.285180+0800 0xde       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.285428+0800 0xe3       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.285556+0800 0xe4       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.285652+0800 0xe1       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.285714+0800 0xe5       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.287093+0800 0xe0       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.291705+0800 0x110      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.301759+0800 0x119      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.302065+0800 0x11a      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.302236+0800 0x11b      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:33.313396+0800 0x122      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:41.498972+0800 0x47d      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:41.500315+0800 0x47e      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:41.500967+0800 0x479      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:39:41.535163+0800 0x483      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:41:34.846164+0800 0x1ea2     Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:41:34.849155+0800 0x1ea5     Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:41:34.849272+0800 0x1ea3     Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:41:34.849455+0800 0x1ea4     Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-12 23:41:34.852124+0800 0x1ead     Default     0x0                  0      0    kernel: (kernel) itlwm: free
这是我的日志

yaozaiyu, 2020-06-12:
感谢，ax200已经工作，100m有20m的速度

M69W, 2020-06-12:
> Greate news! start to scan ssid, but can not connect to my wireless.
> here is the log [itlwmx.log](https://github.com/zxystd/itlwm/files/4771459/itlwmx.log)
> and network setting below.
> [image]

try it that build by @zxystd
[itlwmx_ax200_0612.kext.zip](https://github.com/zxystd/itlwm/files/4772148/itlwmx_ax200_0612.kext.zip)

jianghaizhi, 2020-06-12:
非常感谢，Killer ax1650也能用了

zxystd, 2020-06-13:
> Greate news! start to scan ssid, but can not connect to my wireless.
> here is the log [itlwmx.log](https://github.com/zxystd/itlwm/files/4771459/itlwmx.log)
> and network setting below.
> [image]

according to the log, the card working, scanning but not auth, check your ssid and password configuration

fudaming, 2020-06-13:
> try it that build by @zxystd
> [itlwmx_ax200_0612.kext.zip](https://github.com/zxystd/itlwm/files/4772148/itlwmx_ax200_0612.kext.zip)

[image]

2.4G connected, ready to try 5G

KratosAurion400, 2020-06-13:
@fudaming I can't get the Kext to load it gives me an authentication error even after doing sudo chown -R root:wheel itlwmx.kext did I do something wrong?

yaozaiyu, 2020-06-13:
ax200有时候工作，有时候不工作

zxystd, 2020-06-13:
> @fudaming I can't get the Kext to load it gives me an authentication error even after doing sudo chown -R root:wheel itlwmx.kext did I do something wrong?

Edit the info.plist, add your wifi ssid and password into it. and then put it in clover or opencore kext directory, reboot and enjoy.

zxystd, 2020-06-13:
> ax200有时候工作，有时候不工作

more clearly?

Stevemoretz, 2020-06-13:
> [image]
>
> 2.4G connected, ready to try 5G

Is the upload speed problem solved?

yaozaiyu, 2020-06-13:
> > ax200有时候工作，有时候不工作
> 
> more clearly?

好像是win10重启进入系统就无法驱动

zxystd, 2020-06-13:
> > > ax200有时候工作，有时候不工作
> > 
> > 
> > more clearly?
> 
> 好像是win10重启进入系统就无法驱动

嘤嘤嘤，我的windows系统炸了。不过这是个好消息啊，我终于有设备可以调试热启动的问题了。

fudaming, 2020-06-14:
> @fudaming I can't get the Kext to load it gives me an authentication error even after doing sudo chown -R root:wheel itlwmx.kext did I do something wrong?

as you can find in my log, I run codesign and xattr before kextutil. you can try these command below. If it is not work, please check your Info.plist as @zxystd told.
`
codesign --force --deep --sign - ./itlwmx.kext
xattr -c -r ./itlwmx.kext
chown -R root:wheel ./itlwmx.kext
cp -r ./itlwmx.kext /tmp/
kextutil -v 6 /tmp/itlwmx.kext
`

fudaming, 2020-06-14:
@Stevemoretz 
> Is the upload speed problem solved?

I don' know the upload issues, it seem very stable when I browse the page.

fudaming, 2020-06-14:
I remove the 2.4G SSID in Info.plist, force it connect to my 5G ssid, turns out it will work under the 11n mode. 😂

Anyway, I can remove the Ethernet cable from my computer now.🤪🎉 , thx @zxystd and all the people make it happen.

[image]

yaozaiyu, 2020-06-16:
最近能很好的工作，不管是从win10切换启动还是直接启动，都没有不加载的问题，版本是002淘宝版本，反馈完毕

frankstallone, 2020-06-28:
I got this working, and am grateful! Is it possible to get this to show up as a WiFi adapter? I wanted to get this working, so I could use my Apple Watch to unlock my Hackintosh. I am thinking that is the reason why the option to unlock with Apple Watch is not showing up. 

[image]

closed 2020-07-08 by zxystd

frankstallone, 2020-07-08:
I'll take that as a no. 🤣

BenDevelopment, 2020-08-27:
> 非常感谢，Killer ax1650也能用了

Can you confirm AX1650 is supported? (I cannot find it here https://openintelwireless.github.io/itlwm/Compat.html).
Is AX1650 considered as AX200?

williambj1, 2020-08-27:
@BenDevelopment It's basically a special OEM version of Intel cards with a fancy proprietary driver in Windows. Could you please provide your card's PCI ID?

BenDevelopment, 2020-08-27:
@williambj1 My PCI ID is :

`PCI\VEN_8086&DEV_06F0&SUBSYS_16511A56&REV_00\3&11583659&0&A3`

williambj1, 2020-08-27:
Yes and it's supported