OpenIntelWireless/itlwm #37: 8265ac成功加载驱动并且连接上了网络，但还有一些问题

szhifei, 2020-04-16:
10.15.4的系统，开机速度感觉上没有变慢，刚开机就连接上网络了，只是完全用不了，ping路由器也不行，但是过了一段时间使用百度远景都是秒开，ping百度也只有30多延迟，只看了十几个数据包没有丢包，qq接收发送图片也正常，尝试测速，但是测试不了，又过了一会儿就用不了了，跟刚开机时的表现一样，全程大概二十多分钟
[log.txt](https://github.com/zxystd/itlwm/files/4488552/log.txt)

zxystd, 2020-04-19:
会慢慢解决这些问题

szhifei, 2020-04-20:
还有，电脑连接上wifi后似乎其他设备都无法连接甚至搜索不到wifi，电脑断开后就好了，不知道是不是我的问题
期待正式版驱动

zxystd, 2020-04-20:
路由器型号是啥？

szhifei, 2020-04-20:
用的电信EPON天翼网关的wifi，型号是ZXHN F450，回头我连个路由器试试

szhifei, 2020-04-26:
http://bbs.pcbeta.com/viewthread-1852891-1-1.html

大佬看一下这个帖子，我用了他的套壳驱动，有了wifi管理，也能识别到位置信息了，而且大量上传也没有掉过(用您的最新编译时，qq只发送一张图片都会掉，虽然还连接着，但是用不了)，也没有我之前说的，其他设备搜索不到wifi的问题，只是用了一会儿之后网速变慢，其他没遇到问题，不知道可否有合理解释，希望你们可以联手一下哈哈哈

这是系统报告里的wifi识别信息，还有加载了贴子里楼主的驱动生成的log，是/tmp下加载的
[归档.zip](https://github.com/zxystd/itlwm/files/4535741/default.zip)

zxystd, 2020-05-29:
最新的提交 fbd5b23 已经解决啦。

closed 2020-05-29 by zxystd