OpenIntelWireless/itlwm #38: Network slow and random crash fix possible Timeline

Stevemoretz, 2020-04-19:
Hi I really love this kext.and I know you're in the stage of fixing the above problems but would you please tell us how much do you think it'll take? A couple of days? A couple of months or what?

Thank you so much.

zxystd, 2020-04-19:
No timeline, not a big problem(int code section), I am just using my sparetime to improve it.... There are still so much works need to do although it can work stably. If you can help to solve these problems, just do it.

Stevemoretz, 2020-04-19:
Thank you so much I'm not into kexts I don't know I'm a app developer I can't provide useful help but I will relate to what you just said in a famous topic so people who are capable will come and help.

zxystd, 2020-04-19:
Also a helpful way, thanks. Maybe I should pay some time to write some documents to describe the process/problems/todo-list etc.

Stevemoretz, 2020-04-19:
That's gonna be great I'm sure there are lots of people out there who want to help you. Maybe just in the Readme write about this help wanted situation and provide some ways of communication with you also github has a lot of useful features for continuing and...

jesp010, 2020-04-20:
I appreciate your contribution, please keep working on this project!

0x47726567, 2020-04-20:
@zxystd Having documentation of the project will definitely help some people who want to contribute but it takes a lot of time to write, especially in two languages.

Stevemoretz, 2020-04-20:
It doesn't need to be in two languages,English would do I suppose.

zxystd, 2020-04-22:
Now the network speed is much better, you can try. Upload speed is still slow.

Stevemoretz, 2020-04-22:
Thanks man that's definitely some great news.Can't wait for a release version

zxystd, 2020-05-29:
Now it fix in fbd5b23, thanks.

closed 2020-05-29 by zxystd