OpenIntelWireless/itlwm #40: Use IO80211 instead of Ethernet

zdypk1994, 2020-04-21:
Firstly thank you very much and secondly I would like to ask when the driver will use the IO80211 interface for a true wireless network.

zdypk1994, 2020-04-21:
We need airdrop, hands off, location service, etc.
:P

zxystd, 2020-04-21:
`IO80211Family` is not a open source library and also not publish the SDK. As far as I know, the related projects are just only reverse to be able to use the native WiFi management client so that WiFi can be selected. Airdrop and handoff still need to do a lot of reverse engineering and try, which are not what I need to do at this stage. A network card driver involves a lot of work, which will take a long time for a team to complete, and I am just a person(just spend sparetime).

closed 2020-04-22 by zxystd