OpenIntelWireless/itlwm #41: Unable to build in mojave

guaychou, 2020-04-21:
Hey please help me to build in mojave version

i cannot compile it 

[image]

[image]

zxystd, 2020-04-21:
Change the three target `BaseSDK` to `MacOS` will help you. And here is not suggest to discuss how to build or use it. Thanks.

guaychou, 2020-04-21:
@zxystd i already change it to MacOS and still cannot compiled

[image]

closed 2020-04-22 by zxystd

mojo2012, 2020-05-15:
The problem seems to be that the linux/types.h cannot be found in on of the files. Also Even if you fix this by replacing it with a relative import, it still can't build as fails with this error:
`Assigning to 'bus_addr_t' (aka 'unsigned long long') from incompatible type 'nullptr_t'`