OpenIntelWireless/itlwm #42: 无法手动禁用

kuibug, 2020-04-21:
非常感谢您的付出，9260成功驱动，但是无法通过设置面板和`ifconfig`命令暂停使用该网口使用。
[image]
附上itlwm和对应接口的日志，期待您正式的驱动！
[itlwm.log](https://github.com/zxystd/itlwm/files/4509941/itlwm.log)
[en2.log](https://github.com/zxystd/itlwm/files/4509943/en2.log)
-------
操作系统：10.15.3
网卡型号：Intel 9260ac（PCI ID ：8086:2526）

zxystd, 2020-04-22:
你可以试试最新的，我尝试了命令，可以down掉

kuibug, 2020-04-22:
重新pull编译后仍然无法down掉，是我 itl80211.kext加载失败的原因吗？加载itl80211.kext 时显示依赖错误。
```
/private/tmp/itl80211.kext is invalid; can't resolve dependencies.
/private/tmp/itl80211.kext is invalid; can't resolve dependencies.
/private/tmp/itl80211.kext is invalid; can't resolve dependencies.
```

zxystd, 2020-04-22:
itl80211只是个库，不是驱动，不需要加载。另外我尝试过了是可以的，也可以up

kuibug, 2020-04-22:
好的感谢您的耐心解答，我找别的设备重新尝试一下

closed 2020-04-22 by kuibug