OpenIntelWireless/itlwm #47: how to  use?

RomanceLink, 2020-04-24:
I dont konw  how to use this ,please help me,i download the file，but i dont konw how to make to ac9462 drive!

PeLLmEn, 2020-04-24:
1) You have to download and install xcode in your Mac OS.
2) Run in terminal `git clone https://github.com/zxystd/itlwm.git`
3) Locate to `itlwm` folder and open `itlwm.xcodeproj` with xcode.
4) You have to compile 3 kext `itlwm.kext` , `itlwmx.kext` and `itl80211.kext`. Because i'm not using xcode, I don't know how to get it build all properly, I've just selected settings, that recommended by project, targeted `fw_gen.sh` and sometimes it gives normal results. But i don't understand how does it work, so if somebody can explain i'll be grateful.

zxystd, 2020-04-26:
I will publish the release after I have fix the upload speed issue.

flexwin, 2020-05-01:
when will be publishing the release?

basedtho, 2020-05-02:
i can't load the kexts

```
71 Debug sudo kextload -v itlwm.kext itl80211.kext itlwmx.kext
Requesting load of /Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itlwm.kext.
/Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itlwm.kext failed to load - (libkern/kext) dependency load failed; check the system/kernel logs for errors or try kextutil(8).
Requesting load of /Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itl80211.kext.
/Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itl80211.kext failed to load - (libkern/kext) dependency load failed; check the system/kernel logs for errors or try kextutil(8).
Requesting load of /Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itlwmx.kext.
/Users/tho/Library/Developer/Xcode/DerivedData/itlwm-bwyrhxridbgsnugsfhmjokladmaz/Build/Products/Debug/itlwmx.kext loaded successfully (or already loaded)

```

(built successfully, only had to substitute an ocurrence of nullptr with a 0 in one place)

PeLLmEn, 2020-05-02:
@basedtho don't install it to your system!!!!!! Just use it in opencore (oc/kext) or in clover.

basedtho, 2020-05-02:
it won't connect either way

mehmetyukcell, 2020-05-03:
when will be publishing the release?

mehmetyukcell, 2020-05-03:
> 5\. const char

Where load the kexts.

mehmetyukcell, 2020-05-03:
> * You have to download and install xcode in your Mac OS.
> * Run in terminal `git clone https://github.com/zxystd/itlwm.git`
> * Locate to `itlwm` folder and open `itlwm.xcodeproj` with xcode.
> * In `/itlwm/mac80211.cpp` search for:
>   `    static const char *ssid_name = "ssdt"; static const char *ssid_pwd = "zxyssdt112233";`
>   and change `ssdt` and `zxyssdt112233` to your wifi `SSID` and `password`
> * In `/itlwmx/itlwmx.cpp` search for:
>   `const char *ssid_name = "ssdt"; const char *ssid_pwd = "zxyssdt112233";`
>   and change `ssdt` and `zxyssdt112233` to your wifi `SSID` and `password`
> * After changing you have to compile 3 kext `itlwm.kext` , `itlwmx.kext` and `itl80211.kext`. Because i'm not using xcode, I don't know how to get it build all properly, I've just selected settings, that recommended by project, targeted `fw_gen.sh` and sometimes it gives normal results. But i don't understand how does it work, so if somebody can explain i'll be grateful.
> * I have AC 3168 adapter. I've installed all 3 kext, rebooted and finaly connection worked for me but with very poor speed ~1Mbit/s.

Where load the kexts? Clover EFI or Library extensions

PeLLmEn, 2020-05-03:
Clover or opencore efi

mehmetyukcell, 2020-05-03:
> Clover or opencore efi

Thanks... But, xcode build only one kext. İo80211

Whandcz, 2020-05-04:
> 三叶草或opencore efi
你好  ，最新版我编译时 遇到Build input file cannot be found: '/Users/*/Desktop/itlwm-master/itlwm/FwBinary.cpp'   我折腾了好久也没解决    请问 你能指点我一下怎么解决吗

Roshaanv, 2020-05-07:
> 1. You have to download and install xcode in your Mac OS.
> 2. Run in terminal `git clone https://github.com/zxystd/itlwm.git`
> 3. Locate to `itlwm` folder and open `itlwm.xcodeproj` with xcode.
> 4. In `/itlwm/mac80211.cpp` search for:
>    `    static const char *ssid_name = "ssdt"; static const char *ssid_pwd = "zxyssdt112233";`
>    and change `ssdt` and `zxyssdt112233` to your wifi `SSID` and `password`
> 5. In `/itlwmx/itlwmx.cpp` search for:
>    `const char *ssid_name = "ssdt"; const char *ssid_pwd = "zxyssdt112233";`
>    and change `ssdt` and `zxyssdt112233` to your wifi `SSID` and `password`
> 6. After changing you have to compile 3 kext `itlwm.kext` , `itlwmx.kext` and `itl80211.kext`. Because i'm not using xcode, I don't know how to get it build all properly, I've just selected settings, that recommended by project, targeted `fw_gen.sh` and sometimes it gives normal results. But i don't understand how does it work, so if somebody can explain i'll be grateful.
> 7. I have AC 3168 adapter. I've installed all 3 kext, rebooted and finaly connection worked for me but with very poor speed ~1Mbit/s.

Thank you for your instruction! I followed it, but I only get 2 out of 3 kexts compiled with xcode (itwlmx.kext doesn't get created for me). Any ideas what went wrong?

mehmetyukcell, 2020-05-07:
> Thank you for your instruction! I followed it, but I only get 2 out of 3 kexts compiled with xcode (itwlmx.kext doesn't get created for me). Any ideas what went wrong?

With two kexts— can be tried

zxystd, 2020-05-08:
just select `itlwm` target and compile, and itl80211 and itlwmx are not necessary to use, you don't need to compile or use them.

Roshaanv, 2020-05-08:
> just select `itlwm` target and compile, and itl80211 and itlwmx are not necessary to use, you don't need to compile or use them.

I've now compiled itlwm.kext and put it into 'Clover/kexts/others'. After restarting, it still doesn't work.
Then I tried it with kextload and get "(libkern/kext) failed to load" or sometimes "(libkern/kext) dependency load failed". 

I'm on a Matebook X Pro (2018) with Intel AC 8275 and disabled SIP

Longcoming, 2020-05-08:
Install the kext using hackintool to L/E.

Roshaanv, 2020-05-08:
> Install the kext using hackintool to L/E.

Tried that too, didn't work :/

Drun555, 2020-05-10:
> 1. You have to download and install xcode in your Mac OS.
>
>     2. Run in terminal `git clone https://github.com/zxystd/itlwm.git`
>
>     3. Locate to `itlwm` folder and open `itlwm.xcodeproj` with [...]

Hey! 

Can you tell how you targeted fw_gen.sh, please? I can't compile the kext because I don't have **FwBinary.cpp**, which should be created by **fw_gen.sh**, which is a Build Tool for **fwgen**, but building it don't make any files. I'm completely lost now.

hackslashX, 2020-05-18:
> Hey!
>
>
> Can you tell how you targeted fw_gen.sh, please? I can't compile the kext because I don't have **FwBinary.cpp**, which should be created by **fw_gen.sh**, which is a Build Tool for **fwgen**, [...]

There's a `fw_gen.sh` script in the source. Simply execute it to create the necessary file.

closed 2020-05-29 by zxystd