OpenIntelWireless/itlwm #49: High Sierra support（求加入10.13.6支持）

cyjyyd, 2020-04-26:
希望作者考虑一下兼容10.13.6，N卡用户福音

zxystd, 2020-04-26:
10.13可以用的,我配置的最低是10.13,下载编译就可以了

cyjyyd, 2020-04-28:
10.13.6测试成功，功能目前正常，有时候还是会卡慢

closed 2020-04-28 by cyjyyd

Autines, 2020-05-19:
@zxystd  请问下最新的code是限制了10.15吗？我用10.13的xcode10打不开，提示不兼容。我让别人用10.15编译的版本加载不了，5月份之前的版本是可以用的。

Autines, 2020-05-19:
[image]

williambj1, 2020-05-19:
用命令行编译，工程文件没做向下兼容

Autines, 2020-05-19:
MacBook-Pro:itlwm-master428 autines$ xcodebuild -target itlwm
note: Using new build system
note: Planning build
note: Constructing build description
Build system information
error: unable to find sdk 'macosx10.13' (in target 'itlwm')

Build system information
error: unable to find sdk 'macosx10.13' (in target 'itl80211')

Build system information
warning: The macOS deployment target is set to 10.15, but the range of supported deployment target versions for this platform is 10.6 to 10.14. (in target 'fwgen')

请问这个是缺少10.13的sdk吗？ 
@williambj1

cyjyyd, 2020-05-19:
> MacBook-Pro:itlwm-master428 autines$ xcodebuild -target itlwm
> note: Using new build system
> note: Planning build
> note: Constructing build description
> Build system information
> error: unable to find [...]

是的，请下载一个10.13的SDK，xcode高版本没有内置这个东西

zxystd, 2020-05-19:
你的系统是10.14？xcode默认是只带了当前系统最高版本的sdk，你得用10.13的sdk去编译。另外工程打不开的问题，用记事本打开工程目录下的` itlwm.xcodeproj/project.pbxproj` ，找到` compatibilityVersion= "Xcode 11.0";`，改成你的xcode版本再打开试试。

Autines, 2020-05-19:
我的系统是10.13.6，应该带了10.13的SDK呀

zxystd, 2020-05-19:
那你先按我说的改下工程配置，然后用xcode打开，再改编译sdk版本，记得每个target的sdk版本都要改，包括fw_gen

Autines, 2020-05-19:
compatibilityVersion = "Xcode 10.1";
还是打不开工程文件。

Autines, 2020-05-19:
下载最新的源码，使用xcodebuild -target itlwm编译报错。
The following build commands failed:
	CompileC /Users/autines/Downloads/itlwm-master/build/itlwm.build/Release/itlwm.build/Objects-normal/x86_64/ieee80211_mira.o /Users/autines/Downloads/itlwm-master/itl80211/openbsd/net80211/ieee80211_mira.c normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)