OpenIntelWireless/itlwm #51: Intel 8275: Works but random Kernel Panic

mircoianese, 2020-04-26:
Hello!
First of all thanks for this amazing work. I've tested your kext on my 2018 Matebook X Pro with Intel 8275AC Wireless card. 

It works but I get random Kernel Panic (especially if I do network intensive stuff, like a speedtest).

If you need logs or anything else please let me know. 
Thanks

zxystd, 2020-04-26:
is the master latest code version?

mircoianese, 2020-04-26:
> is the master latest code version?

Actually I'm not totally sure. Could you provide me a compiled kext download link for the last version?
Thanks!

zxystd, 2020-04-26:
[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4535569/itlwm.kext.zip)

Try this one built by latest code, since there is still have upload issue I am trying to fix. But the download speed is fine and would not caused kernel panic I think.

WIFI ssid and password are hardcoded `ssdt` `zxyssdt112233`

mircoianese, 2020-04-26:
I've created a guest network and tried this kext. 

For now it seems to be stable (usually it crashes in less than 3 minutes). I will of course keep testing.
I can confirm the issue with upload speeds.

If you need any other testing done on this card let me know!
Thanks

zxystd, 2020-04-26:
Thanks, If kernel panic happened again, please send me the panic log, it will help me to locate the problem.

mircoianese, 2020-05-01:
I can confirm: no more kernel panics with this kext. I've also setup the environment to compile from source myself - last version I just compiled still has upload speed issues, but everything else seems fine!

Thank you for your work

closed 2020-05-29 by zxystd