OpenIntelWireless/itlwm #52: A weird bug and how it makes the kext works.

marcusbalo, 2020-04-26:
I was trying several methods to get my intel wifi card to work properly, so I found itlwm and [AppleIntelWifi / adapter] (https://github.com/AppleIntelWifi/adapter).
So, as I expected, none of this worked for my 0x3166 wifi card.
The second kext (from the links) loads the firmware and starts correctly, creating a wifi entry in the list of adapters in the network properties. But scan does not work. Can't find networks. Well, with many trials and errors, I found that if I load the first kext, restart the laptop and load the itlwm kext (after editing the ssid and password in xcode), somehow it will make my wifi work. Itlwm creates an Ethernet adapter entry in the network settings, as expected, as it fakes to the system that it is an ethernet adapter. Anyway, it does not work from there, it is still displayed disconnected. BUT, when I look at the wifi adapter entry created earlier by the previous kext, it is connected. lol I can't scan for networks, yet, but it works. The download speed is fine, but the upload is slow and unstable.

How could it be possible ? I was wondering to contribute with the project, but the code looks messy and also the files. Could you explain the structure to make it easier ?

williambj1, 2020-04-26:
@marcusbalo Your network port info is the system is likely messed up, try this:

- Open System Preferences, delete all ports in the Network, click apply in the bottom right corner
- Open Terminal, run: ```open /Library/Preferences/SystemConfiguration```
- Then Delete
  - **`com.apple.airport.preferences.plist`**
  - **`preferences.plist`**
  - **`NetworkInterfaces.plist`**
  - **`com.apple.network.identification.plist`**
  - **`com.apple.wifi.message-tracer.plist`**
  - And all files containing `enX`, (you can find them with grep)
  - (If you've changed SMBIOS before, there will be a file that starts with the models and you must delete them as well)
- Shutdown your device completely, unplug the power cable, wait 5 sec, plug in the power cable, then boot into the system
- NOTE: These files are automatically created, there will be no problems if there are no special settings before, ***backup just in case.***

(I don't want to correct my grammar anymore :p, hope it's easy to understand)

zxystd, 2020-04-26:
Maybe it is not really a bug, reset the network config following https://github.com/zxystd/itlwm/issues/52#issuecomment-619536575 can help you.

There is still exist one big problem is the TX issue, now I even haven't locate how this problem caused, it seems that route not accept the large output packet(maybe packet error or bus transfer error?), and the firmware crash(as shown in network disconnected).

marcusbalo, 2020-04-28:
These methods did not worked. I keep trying everything I can.
It still only works when a load AppleIntelWifi > Reboot > Load Itlwm.
One question, what is the point of itl80211.kext ? It doesn't load (miss dependency)
I verified with kextlibs and this is the output:

```
>  kextlibs -undef-symbols /Users/marcus/Desktop/itl80211.kext

For all architectures:
    No libraries found.

For x86_64:
    2 symbols not found in any library kext:
	__Z6_startP9kmod_infoPv
	__Z5_stopP9kmod_infoPv
```

closed 2020-05-29 by zxystd