OpenIntelWireless/itlwm #53: Hello Intel Wifi Card

tolgasonkaya, 2020-04-26:
Hello I have Intel AC-3165 card. I want to help you with your project to test kexts.
But I do not know how to compile with Xcode as I update the kext here, I want to provide you with information.

mehmetyukcell, 2020-04-26:
> Hello I have Intel AC-3165 card. I want to help you with your project to test kexts.
> 
> But I do not know how to compile with Xcode as I update the kext here, I want to provide you with information.

https://intelwifi.0xcc.pw burda kext var ama Wi-Finizin idsi uyumlu mu bilmiyorum?

tolgasonkaya, 2020-04-26:
Hocam ben osxinfodan TolgaSonn linkteki kext farklı bi projeye ait bu arkadaş ayrı bir kext deniyor çin forumunda konusu var fakat üyelik davetiye sistemiyle ücretli yürüyor

mehmetyukcell, 2020-04-26:
> Hocam ben osxinfodan TolgaSonn linkteki kext farklı bi projeye ait bu arkadaş ayrı bir kext deniyor çin forumunda konusu var fakat üyelik davetiye sistemiyle ücretli yürüyor  

Onu anlayamadım kardeşim ben de burdan kodlarla kext oluşturmaya izin yok mu yani?

zxystd, 2020-04-26:
^qUxJg$c1cdd294dfdd92eb542ad63b164cedc82e2047cdc
thanks, here you can see the binary file i have just compile

tolgasonkaya, 2020-04-27:
Var hocam xcode ile kext oluşturabiliyorsanız sorun yok ben o kadar bilgili değilim

tolgasonkaya, 2020-04-27:
> #51
> thanks, here you can see the binary file i have just compile

Thanks I try kext and send message

tolgasonkaya, 2020-04-27:
Hello there. I load the Kexti L / E directory with Hackintool, but the system does not recognize my wifi card.

Longcoming, 2020-04-27:
Sometimes,there is connection,sometimes there is not.

zxystd, 2020-04-27:
> Hello there. I load the Kexti L / E directory with Hackintool, but the system does not recognize my wifi card.

what is your card? and the product id where you can see in hackintool pci category

tolgasonkaya, 2020-04-27:
> > Hello there. I load the Kexti L / E directory with Hackintool, but the system does not recognize my wifi card.
> 
> what is your card? and the product id where you can see in hackintool pci category

Now installed. id=3165. 
kext sometimes connects sometimes not.

mehmetyukcell, 2020-04-28:
> > > Hello there. I load the Kexti L / E directory with Hackintool, but the system does not recognize my wifi card.
> 
> > 
> 
> > what is your card? and the product id where you can see in hackintool pci category
> 
> 
> 
> Now installed. id=3165. 
> 
> kext sometimes connects sometimes not.
> 
> 

Hacklntool ile wifi bilgilerini paylaş fotoğraf şeklinde.

tolgasonkaya, 2020-04-29:
> Hacklntool ile wifi bilgilerini paylaş fotoğraf şeklinde.

[url=https://hizliresim.com/WoHMSq][img]https://i.hizliresim.com/WoHMSq.png[/img][/url]

closed 2020-05-29 by zxystd