OpenIntelWireless/itlwm #55: i have a 3168 and want to help

balenv, 2020-04-28:
i’ve installed the kext and it works like a charm, it connects to 5g networks without problems. sometimes it disconnects and i have to restart the mac. would be helpful if a send some kind of log or something?

zxystd, 2020-04-28:
Firstly thanks for your feedback, it is an knowned issue and I am finding ways to solve this problem.

PeLLmEn, 2020-04-28:
@zxystd I have 3168 to. Can you give me invitation code to http://bbs.pcbeta.com/ ? it's needed to register and view your thread....

zxystd, 2020-04-28:
> @zxystd I have 3168 to. Can you give me invitation code to http://bbs.pcbeta.com/ ? it's needed to register and view your thread....

Sorry can't help you get the invitation code, but the thread content is almost same as the README.

zxystd, 2020-05-29:
fix in fbd5b23, now upload will not disconnect, and you can unload it without reboot you mac.

closed 2020-05-29 by zxystd

futureorkun, 2020-08-08:
zxystd Can you give me invitation code to http://bbs.pcbeta.com/  Please help me! Thanks...

ErrorErrorError, 2020-08-08:
> me

If it's help on installing you can use gitter: https://gitter.im/OpenIntelWireless/itlwm. Else it's an issue create a new issue request and attach logs https://openintelwireless.github.io/General/Troubleshooting.html.