OpenIntelWireless/itlwm #56: matebook x 2019 AC9560 no lucky

demonlj, 2020-04-28:
osx: 10.15.3
xcode compile OK
kextutil
```
Kext with invalid signature (-67062) allowed: <OSKext 0x7f91c840b150 [0x7fff9518a090]> { URL = "file:///Users/jason/Desktop/itl80211.kext/", ID = "com.zxystd.itl80211" }
/Users/jason/Desktop/itl80211.kext is invalid; can't resolve dependencies.
/Users/jason/Desktop/itl80211.kext is invalid; can't resolve dependencies.
/Users/jason/Desktop/itl80211.kext is invalid; can't resolve dependencies.
Code Signing Failure: not code signed
Validation Failures:
    Info dictionary property value is illegal:
        OSBundleLibraries
    Info dictionary missing required property/value:
        OSBundleLibraries

Diagnostics for /Users/jason/Desktop/itl80211.kext:
Validation Failures:
    Info dictionary property value is illegal:
        OSBundleLibraries
    Info dictionary missing required property/value:
        OSBundleLibraries

Kext with invalid signature (-67062) allowed: <OSKext 0x7f91c8416ab0 [0x7fff9518a090]> { URL = "file:///Users/jason/Desktop/itlwm.kext/", ID = "com.zxystd.itlwm" }
Code Signing Failure: not code signed
```

put itl80211.kext & itlwm.kext into clover/, the log from command
`log show --last boot | grep 'itlwm' >> test.log`

```
2020-04-28 15:13:21.140305+0800 0xfb       Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.140319+0800 0xfd       Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.140358+0800 0xfe       Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.140770+0800 0xfe       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.151456+0800 0xfd       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.181976+0800 0x101      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.182934+0800 0xff       Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.183076+0800 0x10e      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.183081+0800 0x10b      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.183087+0800 0x10a      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.183542+0800 0x107      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.184047+0800 0x10e      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.184062+0800 0xff       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.184062+0800 0x10a      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.184068+0800 0x10b      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.184469+0800 0x100      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.184657+0800 0xfb       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.194556+0800 0x107      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.200815+0800 0x10d      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.201106+0800 0xfc       Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.201109+0800 0x109      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.201111+0800 0x10c      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.201139+0800 0x106      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.205048+0800 0xfc       Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.205066+0800 0x109      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.205068+0800 0x10c      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.205068+0800 0x106      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.205444+0800 0x110      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.215386+0800 0x100      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.225765+0800 0x10d      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.225797+0800 0x110      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.226017+0800 0x104      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.226027+0800 0x108      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.235769+0800 0x108      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.235817+0800 0x104      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:21.246076+0800 0x112      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:21.249842+0800 0x112      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:32.234777+0800 0x4a3      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:32.237813+0800 0x4a3      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:32.239638+0800 0x4ab      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:32.242599+0800 0x49f      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:32.242621+0800 0x49f      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:32.245260+0800 0x4ab      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-04-28 15:13:32.495556+0800 0x49f      Default     0x0                  0      0    kernel: (kernel) itlwm: iwm_match
2020-04-28 15:13:32.501618+0800 0x49f      Default     0x0                  0      0    kernel: (kernel) itlwm: free
```

zxystd, 2020-04-28:
What is your card product id? you can find it by hackintool

mehmetyukcell, 2020-04-28:
> What is your card product id? you can find it by hackintool
I have same problems... product 9560ac id: 8086:0034

zxystd, 2020-04-28:
> > What is your card product id? you can find it by hackintool
> > I have same problems... product 9560ac id: 8086:0034

0034 is not a PCI id, confirm again?

pytason, 2020-04-28:
[image]

I have the same card but I dont get that error

williambj1, 2020-04-28:
Should be  `9DF0`

Ref: https://www.linux-hardware.org/index.php?id=pci:8086-9df0-8086-0034

demonlj, 2020-04-28:
@pytason
also tried kextutil / kextload, not working

pytason, 2020-04-28:
I used this command on the terminal

sudo kextload -v /path to kext/itlwm.kext

demonlj, 2020-04-28:
```
⋊> ~/Desktop on master ⨯ sudo kextload -v itlwm.kext                                           19:31:13
Requesting load of /Users/jason/Desktop/itlwm.kext.
/Users/jason/Desktop/itlwm.kext loaded successfully (or already loaded).
```
kext could be loaded,  eth could also be up, but not fetching an valid ip address
```
en0: flags=863<UP,BROADCAST,SMART,RUNNING,SIMPLEX> mtu 1500
	options=400<CHANNEL_IO>
	ether 11:22:33:44:55:66
	inet 169.254.151.219 netmask 0xffff0000 broadcast 169.254.255.255
```

demonlj, 2020-04-28:
the new log file as below:
```
2020-04-28 19:48:33.283846+0800 0x2ae      Error       0x0                  63     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x6000014c7f60 [0x7fff8bed1090]> { URL = "file:///Users/jason/Sync/itlwm/build/Build/Products/Debug/itlwm.kext/", ID = "com.zxystd.itlwm" }
2020-04-28 19:48:33.477373+0800 0x2ae      Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-04-28 19:48:33.480480+0800 0x52c5     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480490+0800 0x52c5     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.480542+0800 0x52d2     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480562+0800 0x52d2     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.480635+0800 0x52c4     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480643+0800 0x52c4     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.480697+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480755+0800 0x52c3     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480762+0800 0x52c3     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.480949+0800 0x52cd     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.480960+0800 0x52cd     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.481286+0800 0x52c6     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.481296+0800 0x52c6     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.490910+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_attach
2020-04-28 19:48:33.491084+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: , (null)
2020-04-28 19:48:33.491101+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-04-28 19:48:33.491754+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: alloc contig
2020-04-28 19:48:33.505738+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: allocate TX ring
2020-04-28 19:48:33.524049+0800 0x52f4     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.524061+0800 0x52f4     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.524115+0800 0x52f1     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.524124+0800 0x52f1     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.524230+0800 0x52f6     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.524245+0800 0x52f6     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.553422+0800 0x52fc     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.553438+0800 0x52fc     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.593326+0800 0x52fd     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.593348+0800 0x52fd     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.604614+0800 0x52fe     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.604633+0800 0x52fe     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.614398+0800 0x52ff     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match
2020-04-28 19:48:33.614409+0800 0x52ff     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-04-28 19:48:33.633797+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_alloc_rx_ring
2020-04-28 19:48:33.636307+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: config ieee80211
2020-04-28 19:48:33.636310+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) ieee80211_ifattach
2020-04-28 19:48:33.636312+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_channel_init
2020-04-28 19:48:33.636316+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=-1
2020-04-28 19:48:33.636318+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-04-28 19:48:33.636321+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_media_init
2020-04-28 19:48:33.636324+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_preinit
[... cut ...]
2020-04-28 19:48:33.645952+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_start_fw ucode_type=1
2020-04-28 19:48:33.645956+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_apm_init
2020-04-28 19:48:33.645981+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: L1 Disabled - LTR Enabled
2020-04-28 19:48:33.646041+0800 0x52c9     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_load_firmware
```

williambj1, 2020-04-28:
https://github.com/zxystd/itlwm/issues/52#issuecomment-619536575

Follow these procedures and clean up your firmware and network settings, I've experienced the same issue before (log stuck at `iwm_load_firmware`) and I successfully solved it.

demonlj, 2020-04-28:
@williambj1 it works! thank you!

closed 2020-04-28 by demonlj

mehmetyukcell, 2020-04-28:
> > > What is your card product id? you can find it by hackintool
> > > I have same problems... product 9560ac id: 8086:0034
> 
> 0034 is not a PCI id, confirm again or can you provide a screenshot?

[image]

dlhxr, 2020-04-29:
> [#52 (comment)](https://github.com/zxystd/itlwm/issues/52#issuecomment-619536575)
> 
> Follow these procedures and clean up your firmware and network settings, I've experienced the same issue before (log stuck at `iwm_load_firmware`) and I successfully solved it.

Hey, I am using Legion Y540 with Intel AC9560 as well. The solution for log stuck at load firmware is to fully shutdown my laptop for 5 sec. Hearing it's a firmware compatibility issue, is it possible to avoid fully shutdown by using the compatible windows driver? Is there any way to connect Openbsd driver version with windows one's?

Thanks a lot for the 5 sec solution.