OpenIntelWireless/itlwm #61: T480 8265今天更新后使用正常

kai2ou, 2020-04-29:


kai2ou, 2020-04-29:
最大网速20M

ailuoku6, 2020-04-30:
上传也正常了吗

Derekxxzzyy, 2020-04-30:
正常了，就是数据量大会锻断链几秒

kai2ou, 2020-04-30:
> 上传也正常了吗

测试了下，上传速度很慢20k/s以内

91accp, 2020-04-30:
我这边小米笔记本Pro一代  8265AC  上传会直接断开 下载网速大概在15-20M

zxystd, 2020-04-30:
上传有问题，等待修复

mendax955, 2020-04-30:
My laptop is Thinkpad X390 and my wireless wifi is intel ac9560 ,how can l use it on my macOS 10.15.4

kai2ou, 2020-04-30:
> My laptop is Thinkpad X390 and my wireless wifi is intel ac9560 ,how can l use it on my macOS 10.15.4

Add the kext files to your EFI, change your wifi name and pwd to "ssd" and "zxyssdt112233"

> https://kai2ou.lanzous.com/b0aetj6fe 
密码:aks7

mendax955, 2020-05-01:
Thank you ,but l think a565109863's wifi kext is better than zxystd's itlwm. Although he has copied zxystd's itlwm ,he is still very great because his work is nothing but splendid

lulujyc, 2020-05-01:
> Thank you ,but l think a565109863's wifi kext is better than zxystd's itlwm. Although he has copied zxystd's itlwm ,he is still very great because his work is nothing but splendid

Respect GPL.

gplast, 2020-05-01:
@mendax955 This is an effort of @zxystd to do something great for the Hackinoths community and you should respect that! The project is a work in progress and you should be grateful for all the effort that this man gives for us without even getting paid! 

@zxystd keep up the great work and i suggest you add also a donate link on the repository to let people support your effort and time that you spend on that.

You effort is really appreciated and thank you!

mendax955, 2020-05-02:
@gplast @lulujyc  Yeah,l will really agree with you and l absolutely respect GPL.However, from my perspective, this is just a small comparison between these two donors.And l really hope zxystd to learn a little from a565109863's wifi kext(l know a565109863's has copied some part of zxystd's contribution, but a565109863's wifi kext still has many respects to be learned).Although he didn't respect GPL, in order to let the wifi kext to be finally developed, don't ban other's great contribution because of his ignorance of GPL.His kext is really awesome and so does zxystd's. Let's make the itlwm's world better and let's fight together!

Goshin, 2020-05-02:
> @gplast @lulujyc Yeah,l will really agree with you and l absolutely respect GPL.However, from my perspective, this is just a small comparison between these two donors.And l really hope zxystd to [...]

Actually no one has ever banned his "great contribution". He is always free to turn his project private if he makes use of other's GPL source code and doesn't want to release his work under the GPL license. 
And what kind of "learning" are you proposing for a closed-source work? 🤔 confused.

closed 2020-05-29 by zxystd