OpenIntelWireless/itlwm #62: iwm_rx_tx_cmd_single 613 OUTPUT_ERROR status=132

s0ullight, 2020-04-29:
[log.txt](https://github.com/zxystd/itlwm/files/4555116/log.txt)

The 3160 firmware gives a fatal error right after iwm_rx_tx_cmd_single 613 OUTPUT_ERROR status=132.
It happens twice in the attached log.
On a side note: unloading the kext causes system crash

zxystd, 2020-04-30:
Yes, it is still a tx issue, just because of the packet error or bus transfer wrong(not sure), then the router refuse the packet. really a strange problem, the sequence is just convert frame to 802.11 which from `outputPacket`, and do bus send through `iwm_tx` (even non encrypted WiFi also contains this problem, so we can eliminate the problem of algorithm).Network down, firmware crash, slow upload speed, unstable ping stat... these problems are the same cause of this.

mehmetyukcell, 2020-05-01:
> Yes, it is still a tx issue, just because of the packet error or bus transfer wrong(not sure), then the router refuse the packet. really a strange problem, the sequence is just convert frame to [...]

When can the kext command come out?

s0ullight, 2020-05-29:
Commit 81 fixed this crash

closed 2020-05-29 by s0ullight