OpenIntelWireless/itlwm #64: Will this kext support airport?

Mr-UFO, 2020-05-02:
Firstly, thanks a lot for your great work!
I am considering BCM943602CS and BCM94352Z just only for they support airport, however they are much more expensive than Intel wireless card.
If there is any possibility of being driven, I will wait until you make it!

maz-1, 2020-05-02:
The driver is not using IO80211Family API, so probably not.

zxystd, 2020-05-04:
Everything is possible and needs time to develop. This project isn't a replacement of Broadcom  or native network cards, it is just a choice for those who do not want to change their cards.

Mr-UFO, 2020-05-04:
@zxystd 
Thanks for your great efforts and contributions!
You have broken through public perception!
Wish you success!

closed 2020-05-29 by zxystd