OpenIntelWireless/itlwm #65: compile error

demonlj, 2020-05-04:
osx: 10.15.3
xcode: 11.4.1
code: latest commit:883354164327c76825ab5fe89e8ea9eb28473f36
```
note: Using new build system
note: Building targets in parallel
note: Planning build
note: Constructing build description
error: No signing certificate "Mac Development" found: No "Mac Development" signing certificate matching team ID "YXX7XALBVX" with a private key was found. (in target 'itl80211' from project 'itlwm')

** BUILD FAILED **
```

mehmetyukcell, 2020-05-04:
> osx: 10.15.3
>
> xcode: 11.4.1
>
> code: latest commit:883354164327c76825ab5fe89e8ea9eb28473f36
>
> ```
>
> note: Using new build system
>
> note: Building targets in parallel
>
> note: Planning build
>
> note: [...]

Please Xcode prefrences Apple İd sign in

demonlj, 2020-05-04:
already sign in with apple id and apple development certificates, still got errors
```
error: No account for team "YXX7XALBVX". Add a new account in the Accounts preference pane or verify that your accounts have valid credentials. (in target 'itl80211' from project 'itlwm')
error: No signing certificate "Mac Development" found: No "Mac Development" signing certificate matching team ID "YXX7XALBVX" with a private key was found. (in target 'itl80211' from project 'itlwm')
```

zxystd, 2020-05-04:
Just remove the signature of the three target and compile again.
[image]

demonlj, 2020-05-04:
it works. Thanks

closed 2020-05-04 by demonlj