OpenIntelWireless/itlwm #69: rateset函数错误

Cao-czy, 2020-05-04:
/itl80211/openbsd/net80211/ieee80211_mira.c    170

[image]

编译时会因为没有返回报错，我返回了一个false后编译成功并且成功驱动我的8265，并在连接成功后卡死，当然卡死是因为我返回false导致的，但是我去查看了这个函数中的数组ieee80211_std_ratesets_11n，我的AP速率是存在其中的，但是不知道为什么没有正确返回rs以至于返回了我写的false，我不是很敢随意指定一个rs来返回，那样不同的网卡没法通用，想问问这段代码的编译报错有没有解决方案，如果确实没有找到符合的rs后续会如何呢，毕竟我编译如果没有返回就会直接不过，我还看到代码中有AC的速率表，但是关于AC的rateset我只在ieee80211_node.h的135行发现了
`extern const struct ieee80211_vht_rateset ieee80211_std_ratesets_11ac[];`
不知道2.4G与5G的rateset实现方式有什么不同。

zxystd, 2020-05-08:
这个报错跟工程配置有关系，itlwmx那个项目可以不用编译。另外这个80211不支持11ac，也没有真正完整支持11n，我在尝试更新80211部分

Cao-czy, 2020-05-08:
我知道itlwmx是你自己的版本，我没有用那个，那个在我机器上起不来，那应该如何配置工程才能避免这个报错呢

Cao-czy, 2020-05-08:
解决了，谢谢

Cao-czy, 2020-05-08:
尝试了很多方案这个报错依然存在，修改这个文件类型为C Source可以编译成功，但是驱动起不来，看起来必须要Objective-C，也尝试过修改编译器Apple Clang的配置，这个报错依然无解

zxystd, 2020-05-08:
xcode版本？

Cao-czy, 2020-05-08:
mac是13，Xcode最大10.1，我改成10.0compatible试过了

kai2ou, 2020-05-10:
185行添加一行`return NULL;`

Cao-czy, 2020-05-10:
> 185行添加一行`return NULL;`

woc我怎么没想到，我返回个false肯定炸啊，傻了傻了，万分感谢，已经成功连接。

closed 2020-05-29 by zxystd