OpenIntelWireless/itlwm #71: 有计划支持wpa2-enterprise 吗？

charlesld, 2020-05-08:
wpa2下认证正常使用，wpa2企业级无法使用。  

[image]

特来咨询下作者，是否有支持计划？

[image]

charlesld, 2020-05-08:
另，加入的网络wpa2会被误认为wep
[image]

zxystd, 2020-05-08:
额，你应该是用的别人的kext。

closed 2020-05-08 by zxystd

WarrenChan1027, 2020-08-03:
所以现在可以支持了，公司的wifi我连不了。。

WarrenChan1027, 2020-08-03:
所以现在可以支持了吗？公司的WiFi我还是连不了。

charlesld, 2020-08-03:
@WarrenChan1027  我是买了个网卡，能用公司的网络，intel的只能用家里的。

WarrenChan1027, 2020-08-03:
是买的USB无线网卡吗

charlesld, 2020-08-03:
@WarrenChan1027  是的。 https://www.jianshu.com/p/964217d72ef1

WarrenChan1027, 2020-08-04:
不好意思，你发的这个简书的链接我打不开。。你能发个具体的型号吗

charlesld, 2020-08-04:
@WarrenChan1027  CF-812AC