OpenIntelWireless/itlwm #72: Crash on OpenBSD firmware load

dohnuts, 2020-05-08:
```
iwx0: dumping device error log
iwx0: Start Error Log Dump:
iwx0: Status: 0x1, count: 6
iwx0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL
iwx0: 002022F0 | trm_hw_status0
iwx0: 00000000 | trm_hw_status1
iwx0: 004FC308 | branchlink2
iwx0: 00016E5A | interruptlink1
iwx0: 00016E5A | interruptlink2
iwx0: 004F9F62 | data1
iwx0: 00001000 | data2
iwx0: F0000008 | data3
iwx0: 00000000 | beacon time
iwx0: 00011B6F | tsf low
iwx0: 00000000 | tsf hi
iwx0: 00000000 | time gp1
iwx0: 00011B6F | time gp2
iwx0: 00000001 | uCode revision type
iwx0: 0000002E | uCode version major
iwx0: 177B3E46 | uCode version minor
iwx0: 00000340 | hw version
iwx0: 00889000 | board version
iwx0: 800AFD0C | hcmd
iwx0: 00020000 | isr0
iwx0: 00000000 | isr1
iwx0: 18F00002 | isr2
iwx0: 04C0000C | isr3
iwx0: 00000000 | isr4
iwx0: 00000000 | last cmd Id
iwx0: 004F9F62 | wait_event
iwx0: 00000000 | l2p_control
iwx0: 00000020 | l2p_duration
iwx0: 00000000 | l2p_mhvalid
iwx0: 00000000 | l2p_addr_match
iwx0: 00000009 | lmpm_pmg_sel
iwx0: 19071335 | timestamp
iwx0: 00000024 | flow_handler
iwx0: Start UMAC Error Log Dump:
iwx0: Status: 0x1, count: 7
iwx0: 0x201010A3 | ADVANCED_SYSASSERT
iwx0: 0x00000000 | umac branchlink1
iwx0: 0xC008B1C0 | umac branchlink2
iwx0: 0xC0084E04 | umac interruptlink1
iwx0: 0x00000000 | umac interruptlink2
iwx0: 0x00000002 | umac data1
iwx0: 0x00000001 | umac data2
iwx0: 0xDEADBEEF | umac data3
iwx0: 0x0000002E | umac major
iwx0: 0x177B3E46 | umac minor
iwx0: 0x00011B60 | frame pointer
iwx0: 0xC0886C6C | stack pointer
iwx0: 0x00050088 | last host cmd
iwx0: 0x00000000 | isr status reg
driver status:
  tx ring  0: qid=0  cur=5   queued=0
  tx ring  1: qid=1  cur=0   queued=0
  tx ring  2: qid=2  cur=0   queued=0
  tx ring  3: qid=3  cur=0   queued=0
  tx ring  4: qid=4  cur=0   queued=0
  tx ring  5: qid=5  cur=0   queued=0
  tx ring  6: qid=6  cur=0   queued=0
  tx ring  7: qid=7  cur=0   queued=0
  tx ring  8: qid=8  cur=0   queued=0
  tx ring  9: qid=9  cur=0   queued=0
  tx ring 10: qid=10 cur=0   queued=0
  tx ring 11: qid=11 cur=0   queued=0
  tx ring 12: qid=12 cur=0   queued=0
  tx ring 13: qid=13 cur=0   queued=0
  tx ring 14: qid=14 cur=0   queued=0
  tx ring 15: qid=15 cur=0   queued=0
  tx ring 16: qid=16 cur=0   queued=0
  tx ring 17: qid=17 cur=0   queued=0
  tx ring 18: qid=18 cur=0   queued=0
  tx ring 19: qid=19 cur=0   queued=0
  tx ring 20: qid=20 cur=0   queued=0
  tx ring 21: qid=21 cur=0   queued=0
  tx ring 22: qid=22 cur=0   queued=0
  tx ring 23: qid=23 cur=0   queued=0
  tx ring 24: qid=24 cur=0   queued=0
  tx ring 25: qid=25 cur=0   queued=0
  tx ring 26: qid=26 cur=0   queued=0
  tx ring 27: qid=27 cur=0   queued=0
  tx ring 28: qid=28 cur=0   queued=0
  tx ring 29: qid=29 cur=0   queued=0
  tx ring 30: qid=30 cur=0   queued=0
  rx ring: cur=263
  802.11 state INIT
iwx0: fatal firmware error

```

how to help fix ?

zxystd, 2020-05-16:
This is a headache for me right now. Firmware will crash when data is uploaded in large traffic. Besides, everything is well. 
https://github.com/zxystd/itlwm/projects/1#card-37471552

zxystd, 2020-05-29:
duplicate with #72, will support soon.

closed 2020-05-29 by zxystd