OpenIntelWireless/itlwm #76: 3165 Can't load kext com.zxystd.itlwm - link failed. ？？

RANSAA, 2020-05-15:
2020-05-15 12:55:08.941834+0800 0xc7       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV13IOEventSource' for vtable '__ZTV11IOTaskQueue' is out of date. Make sure your kext has been built against the correct headers.
2020-05-15 12:55:08.941843+0800 0xc7       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV15pci_intr_handle' is out of date. Make sure your kext has been built against the correct headers.
2020-05-15 12:55:08.941852+0800 0xc7       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV20IOEthernetController' for vtable '__ZTV5itlwm' is out of date. Make sure your kext has been built against the correct headers.
2020-05-15 12:55:08.941860+0800 0xc7       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV8CTimeout' is out of date. Make sure your kext has been built against the correct headers.
2020-05-15 12:55:08.943623+0800 0xc7       Default     0x0                  0      0    kernel: Can't load kext com.zxystd.itlwm - link failed.
2020-05-15 12:55:08.943756+0800 0xc7       Default     0x0                  0      0    kernel: Failed to load executable for kext com.zxystd.itlwm.
2020-05-15 12:55:08.943781+0800 0xc7       Default     0x0                  0      0    kernel: Kext com.zxystd.itlwm failed to load (0xdc008016).
2020-05-15 12:55:08.943784+0800 0xc7       Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.itlwm (error 0xdc008016).
2020-05-15 12:55:08.945037+0800 0xc7       Default     0x0                  0      0    kernel: Couldn't alloc class "itlwm"
2020-05-15 12:55:16.151683+0800 0x255      Error       0x0                  43     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext id com.zxystd.itlwm not found; removing personalities from kernel.
[log.txt](https://github.com/zxystd/itlwm/files/4632320/log.txt)

zxystd, 2020-05-16:
Seems you are using a wrong compile version, what is your OS version? Or you can try to compile self and run it.

RANSAA, 2020-05-18:
> Seems you are using a wrong compile version, what is your OS version? Or you can try to compile self and run it.

environment：Clover + Mac OSX 10.14.6 + Xcode11.3.1
other：Already exists USB Realtek
Explanation：Pull the latest code to compile, there is still this error after loading kext
[log.txt](https://github.com/zxystd/itlwm/files/4642254/log.txt)

zxystd, 2020-05-18:
try to rebuild your kernel cache.

RANSAA, 2020-05-18:
> try to rebuild your kernel cache.

Still not working

zxystd, 2020-05-18:
Would you mind upload your compile binary file to me?

RANSAA, 2020-05-19:
> Would you mind upload your compiled binary file to me?

[Debug.zip](https://github.com/zxystd/itlwm/files/4648418/Debug.zip)

cmvee, 2020-05-26:
Having a very similar issue. 10.14.6, latest security updates. OpenCore 0.5.8. Intel 9560.

Xcode builds fine. But loading the extension fails with a linker error.

Output of `sudo kextload -v itlwm.kext`:

>(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV13IOEventSource' for vtable '__ZTV11IOTaskQueue' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV15pci_intr_handle' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV20IOEthernetController' for vtable '__ZTV5itlwm' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV8CTimeout' is out of date. Make sure your kext has been built against the correct headers.
(kernel) Can't load kext com.zxystd.itlwm - link failed.
(kernel) Failed to load executable for kext com.zxystd.itlwm.
(kernel) Kext com.zxystd.itlwm failed to load (0xdc008016).
(kernel) Failed to load kext com.zxystd.itlwm (error 0xdc008016).
Failed to load /Users/chris/Library/Developer/Xcode/DerivedData/itlwm-csqzfcedfmmwtffodbmbrhkfebru/Build/Products/Debug/itlwm.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

Output of `sudo kextlibs -undef-symbols itlwm.kext`:

>For all architectures:
    com.apple.iokit.IONetworkingFamily = 3.4
    com.apple.iokit.IOPCIFamily = 2.9
    com.apple.kpi.bsd = 18.7
    com.apple.kpi.iokit = 18.7
    com.apple.kpi.libkern = 18.7
    com.apple.kpi.mach = 18.7
>
>For x86_64:
    7 symbols not found in any library kext:
	__ZN19IONetworkController29_RESERVEDIONetworkController5Ev
	__ZN19IONetworkController29_RESERVEDIONetworkController4Ev
	__ZN19IONetworkController29_RESERVEDIONetworkController3Ev
	__ZN19IONetworkController29_RESERVEDIONetworkController2Ev
	__ZN8OSObject8DispatchE5IORPC
	__ZN9IOService8DispatchE5IORPC
	__ZN15OSMetaClassBase8DispatchE5IORPC

zxystd, 2020-05-29:
Compile issue, will release binary file in release. thanks.

closed 2020-05-29 by zxystd

mackonsti, 2020-07-25:
Hi guys, I faced the same issue, on a locally compiled MacOS Mojave with Xcode 9.4.1, the kext compiles fine with adding the needed headers to Xcode (not sure why):

```
$ sudo installer -pkg /Library/Developer/CommandLineTools/Packages/macOS_SDK_headers_for_macOS_10.14.pkg -target /
```
But pre-compiled kext fails as I have reported here https://github.com/1hbb/OpenIntelWireless-Factory/issues/1

zxystd, 2020-07-25:
@mackonsti You can ask there, BTW, you can try this one, support 10.12+.
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4975217/itlwm.kext.zip)

mackonsti, 2020-07-25:
Thank you @zxystd ! It works! What have you changed/improved? I am very happy!
```
$ sudo kextutil -v itlwm.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-67062) allowed: <OSKext 0x7fab89c06f90 [0x7fff90e968f0]> { URL = "file:///Users/mackonsti/Download/itlwm.kext/", ID = "com.zxystd.itlwm" }
Code Signing Failure: not code signed
/Users/mackonsti/Download/itlwm.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/mackonsti/Download/itlwm.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
/Users/mackonsti/Download/itlwm.kext successfully loaded (or already loaded).
```
Will this fix be available permanently in your official releases from now on?

Thanks for taking time to fix compatibility mate, appreciate it! Hope you've also included some versioning so we know what is old and what is new kext, too! Have a good day.

zxystd, 2020-07-25:
@mackonsti Nothing need to be optimize, just use 10.12 xcode sdk to compile. public release and pre-release will also support 10.12+.
[image]

mackonsti, 2020-07-25:
Looking forward to your official releases @zxystd Xie Xie