OpenIntelWireless/itlwm #77: AC9462 issue

JulioCMLP, 2020-05-15:
I had compiled the source code, I got itlwm and itl80211 I put them in EFI clover but it doesn't work, my chipset is not recognized

kai2ou, 2020-05-15:
ssid_name and said_pwd is ok?

JulioCMLP, 2020-05-15:
Yes it is

kai2ou, 2020-05-15:
Try [https://kai2ou.lanzous.com/icesejc](url), 

> Wi-Fi SSID is hardcoded as ssdt password as zxyssdt112233 and will auto-connect once the Kext is loaded.

JulioCMLP, 2020-05-16:
But my SSID is not that and I can't access to my router coz I don't have the password

JulioCMLP, 2020-05-16:
In my router there's a password but I can't access coz it say wrong password

kai2ou, 2020-05-16:
> But my SSID is not that and I can't access to my router coz I don't have the password

What's your wifi name and password? I will compile a new kext for you

kai2ou, 2020-05-16:
Or you can change the wifi name and password in itlwm/mac80211.cpp

JulioCMLP, 2020-05-16:
MOVISTAR_8400 
mG7JtZx7dZkDYscZt6zp

Please compile a kext, I will be really grateful

kai2ou, 2020-05-16:
> MOVISTAR_8400
> mG7JtZx7dZkDYscZt6zp
> 
> Please compile a kext, I will be really grateful

https://kai2ou.lanzous.com/icoid2j

JulioCMLP, 2020-05-16:
I will prove it tomorrow then I told you if it's work, thank you

JulioCMLP, 2020-05-16:
Should I put it in C/K/O or L/E?

kai2ou, 2020-05-16:
> C/K/O
C/K/O, then restart

JulioCMLP, 2020-05-16:
Hi I used the kext, now my wifi adapter is recognized but never connect to my WiFi, I checked my SSID and PWD and it's correct but don't connect to WiFi, plus I was checking in network preferences and doesn't appear WiFi

JulioCMLP, 2020-05-16:
Forget what I said before, now it's working I just removed all network interface then L/P/SC I removed all the files then reboot and it's worked. In network preferences say Ethernet connected but not wifi I don't know why;  it's working now but speed is slow

JulioCMLP, 2020-05-16:
@kai2ou can You help me how to compile the source code? I don't have any idea about XCode

arch-dev, 2020-05-16:
> @kai2ou can You help me how to compile the source code? I don't have any idea about XCode

Just open itlwm.xcodeproj with xcode and click on build (play button on the top left of the program). When build finish in the project tree (click on first icon under build button, should be a folder icon) scroll down to "Products", right click on it and click on "Show in Finder" here you should have the kext you have just compiled.

JulioCMLP, 2020-05-16:
I get an error about FwBinary

arch-dev, 2020-05-16:
What kind of error?

arch-dev, 2020-05-16:
> Forget what I said before, now it's working I just removed all network interface then L/P/SC I removed all the files then reboot and it's worked. In network preferences say Ethernet connected but not wifi I don't know why; it's working now but speed is slow

You see ethernet because this kext use other drivers that make it recognized by Mac OS as ethernet and not as WiFi but it remains WiFi under the hood (@zxystd correct me if I'm wrong)

JulioCMLP, 2020-05-16:
@arch-dev it say I need FwBinary.cpp

arch-dev, 2020-05-16:
Are you using last sources?

JulioCMLP, 2020-05-16:
Yep, I think I need to generate the FwBinary.cpp running the sh file, but I don't know how to run it

JulioCMLP, 2020-05-16:
I get the same error of this #67

arch-dev, 2020-05-16:
Ok try the following:
1) Close project from xcode
2) Delete sources
3) On xcode startup screen select "Clone repository" (or something like that) and insert link to GitHub repo

JulioCMLP, 2020-05-16:
My problem is how to run the SH script to generate the FwBinary.cpp

arch-dev, 2020-05-16:
It should be run by xcode when building try what I said before

JulioCMLP, 2020-05-17:
[image]
I got that error @arch-dev

JulioCMLP, 2020-05-17:
@arch-dev forget it, already working I compiled by myself now with the latest source code

arch-dev, 2020-05-17:
👍

tolgasonkaya, 2020-05-17:
Can you teach me how to compile with XCode, please?

arch-dev, 2020-05-17:
Look here https://github.com/zxystd/itlwm/issues/77#issuecomment-629668695

tolgasonkaya, 2020-05-18:
> > @ kai2ou Kaynak kodunu derlememde bana yardımcı olabilir misin? XCode hakkında hiçbir fikrim yok
> 
> Sadece itlwm.xcodeproj dosyasını xcode ile açın ve oluştur'a tıklayın (programın sol üst köşesindeki yürüt düğmesi). Proje ağacında derleme bittiğinde (derleme düğmesinin altındaki ilk simgeye tıklayın, bir klasör simgesi olmalıdır) "Ürünler" e gidin, sağ tıklayın ve "Finder'da Göster" e tıklayın burada sadece sahip olduğunuz kext'e sahip olmalısınız derlenmiş.

> Just open itlwm.xcodeproj with xcode and click on build (play button on the top left of the program). When build finish in the project tree (click on first icon under build button, should be a [...]

After compiling the project, I go to the products section, but there is no show button in Finder, I can add videos.

tvhung83, 2020-05-18:
To compile the code, I usually do following:
1. Remove Team in XCode settings
2. Compile & load `itlwm.kext` (you don't need to compile or load other scheme/target)
```shell
$ mkdir build
$ xcodebuild -scheme itlwm -configuration Debug -derivedDataPath build
$ cd build/Build/Products/Debug
$ sudo chown -R root:wheel itlwm.kext
$ sudo kextload -v 6 itlwm.kext
```
I don't think it's stable enough to put it to EFI bootloader.

williambj1, 2020-05-18:
> To compile the code, I usually do following:
>
> 1. Remove Team in XCode settings
> 2. Compile & load `itlwm.kext` (you don't need to compile or load other scheme/target)
>
> ```shell
> $ mkdir build
> $ [...]

@tvhung83  adding `CODE_SIGN_IDENTITY="" CODE_SIGNING_REQUIRED=NO` as a part of the `xcodebuild` command argument should do the same as your first step.

But still it's not recommended for people entirely new with Xcode to try messing up with this kext, it's better to read a rigid tutorial and learn some basics first.

For those who are throwing this kext into L/E even S/L/E: GOOD LUCK

bangdc90, 2020-05-19:
@williambj1 , @tvhung83 
I want to debug with this kext, but I don't know where to check log, could you help me or give me any guideline link. THank you.

williambj1, 2020-05-19:
use `log` and `grep` the keyword "itlwm" should be enough, the `contains` argument in `log` may perform better, the final result of these two filters are the same.

I'd suggest reading ```man log``` and ```man grep``` first, and google the terms you don't understand.

P.S. You can use macOS' builtin console as well, but it's **EXTREMELY BUGGY and LAGGY**.

bangdc90, 2020-05-19:
@williambj1 
Thanks. You mean after kextload, I will open other terminal and run bellow command:
$log | grep "itlwm"

JulioCMLP, 2020-05-24:
@zxystd Hey you ok? 

I'm using latest source code and sometimes my system get freezed then it reboot in kernel panic is maybe because any error on the code?

zxystd, 2020-05-24:
> @zxystd Hey you ok?
> 
> I'm using latest source code and sometimes my system get freezed then it reboot in kernel panic is maybe because any error on the code?

I can't answer without any panic log.

JulioCMLP, 2020-05-24:
> > @zxystd Hey you ok?
> > I'm using latest source code and sometimes my system get freezed then it reboot in kernel panic is maybe because any error on the code?
> 
> I can't answer without any panic log.

How can I get it?

arch-dev, 2020-05-24:
> > > @zxystd Hey you ok?
> > > I'm using latest source code and sometimes my system get freezed then it reboot in kernel panic is maybe because any error on the code?
> > 
> > 
> > I can't answer without any panic log.
> 
> How can I get it?

Panic logs should be under /Library/Logs/...Diagnostics... (Or something like that)

JulioCMLP, 2020-05-24:
[
[Kernel_2020-05-24-155142_MacBook-Pro.txt](https://github.com/zxystd/itlwm/files/4674400/Kernel_2020-05-24-155142_MacBook-Pro.txt)

](url)

JulioCMLP, 2020-05-25:
@zxystd Can you please check it? I'm still having it randomly. I'm waiting your reply

zxystd, 2020-05-25:
The log contains anything related to the kext, did you upload the right one?

JulioCMLP, 2020-05-25:
@zxystd yes it's the latest log I have in that moment comparing with the time... If there's nothing so I suppose the kext isn't the problem

closed 2020-05-29 by zxystd