OpenIntelWireless/itlwm #78: Trouble with fw_gen.sh

awsmct, 2020-05-15:
When i try run fw_gen.sh i get this error:
./fw_gen.sh: line 11: [[!-f: command not found

Last master branch, what i do wrong?

arch-dev, 2020-05-16:
Hi. Edit line 11 as follows:
[[ ! -f
Spaces are important else it will not work.

ksh1t1z, 2020-05-16:
Hi, I'm getting this error after putting in spaces.

line 11: syntax error in conditional expression: unexpected token `;'
line 11: syntax error near `;'
line 11: `if [[ ! -f $target_file]]; then'

awsmct, 2020-05-16:
ur line should looks like "if [[ ! -f $target_file ]]; then"

awsmct, 2020-05-16:
> Hi. Edit line 11 as follows:
> [[ ! -f
> Spaces are important else it will not work.

I need help again.
After start fw_gen.sh i get next errors:

./fw_gen.sh: line 13: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 14: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 20: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 21: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 22: /itlwm/firmware/*: No such file or directory
./fw_gen.sh: line 23: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 24: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 25: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 28: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 30: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 36: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 39: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 40: /itlwm/FwBinary.cpp: No such file or directory

arch-dev, 2020-05-16:
Add this line just before "target_file=...":
PROJECT_DIR="."

awsmct, 2020-05-16:
> PROJECT_DIR="."

Thanks you, man.
If you are dev, fix this on branch, please.

arch-dev, 2020-05-16:
I'm not dev of this project, sorry. @zxystd is dev and I think he will fix this shortly

ksh1t1z, 2020-05-16:
can you create a pull request fixing all these

ksh1t1z, 2020-05-16:
What did I do wrong?

line 9: ${"PROJECT_DIR"}/itlwm/FwBinary.cpp: bad substitution
line 14: $target_file: ambiguous redirect
line 15: $target_file: ambiguous redirect
line 21: $target_file: ambiguous redirectline 22: $target_file: ambiguous redirect
line 23: ./itlwm/firmware/*: No such file or directory
line 24: $target_file: ambiguous redirect
line 25: $target_file: ambiguous redirect
line 26: $target_file: ambiguous redirect
line 31: $target_file: ambiguous redirect
line 37: $target_file: ambiguous redirect
line 40: $target_file: ambiguous redirect
line 41: $target_file: ambiguous redirect

arch-dev, 2020-05-16:
Send your fw_gen.sh so I can check it please

ksh1t1z, 2020-05-16:
Sorry for the trouble.
#!/bin/sh

#  fw_gen.sh
#  itlwm
#
#  Created by qcwap on 2020/3/10.
#  Copyright © 2020 钟先耀. All rights reserved.
PROJECT_DIR="."
target_file="${"PROJECT_DIR"}/itlwm/FwBinary.cpp"
fw_files=${PROJECT_DIR}/itlwm/firmware/*

if [[ ! -f $target_file ]]; then

echo "//  itlwm\n\n//  Copyright © 2020 钟先耀. All rights reserved." >$target_file
echo "#include \"FwData.h\"">>$target_file

for fw in $fw_files; do
    fw_file_name=`basename $fw`
    fw_var_name=${fw_file_name//./_}
    fw_var_name=${fw_var_name//-/_}
    echo "">>$target_file
    echo "const unsigned char ${fw_var_name}[] = {">>$target_file
    xxd -i <$fw >>$target_file
    echo "};">>$target_file
    echo "">>$target_file
    echo "const long int ${fw_var_name}_size = sizeof(${fw_var_name});">>$target_file
done

echo "">>$target_file

echo "const struct FwDesc fwList[] = {">>$target_file
i=0;
for fw in $fw_files; do
    fw_file_name=`basename $fw`
    fw_var_name=${fw_file_name//./_}
    fw_var_name=${fw_var_name//-/_}
    echo "{IWL_FW(\"$fw_file_name\", $fw_var_name, ${fw_var_name}_size)},">>$target_file
    let i+=1
done
echo "};">>$target_file
echo "const int fwNumber = $i;">>$target_file

fi

arch-dev, 2020-05-16:
Change this line
target_file="${"PROJECT_DIR"}/itlwm/FwBinary.cpp"
to this
target_file="${PROJECT_DIR}/itlwm/FwBinary.cpp"

zxystd, 2020-05-16:
I think I have fix this problem, try using the latest commit? thanks for your attention.

awsmct, 2020-05-16:
> I think I have fix this problem, try using the latest commit? thanks for your attention.

Say please, where i can find .kext after build? I can't find it

ksh1t1z, 2020-05-16:
~/Library/Developer/Xcode/DerivedData

zxystd, 2020-05-16:
If ur using the latest commit, you can get it on the projectpath/Build/Products/Debug/itlwm.kext

awsmct, 2020-05-16:
nothing in projectpath and in Xcode/DerivedData

kai2ou, 2020-05-16:
I just tried it, and it is fine.

ksh1t1z, 2020-05-16:
I'm still getting this
[image]

zxystd, 2020-05-16:
just build again

awsmct, 2020-05-16:
> If ur using the latest commit, you can get it on the projectpath/Build/Products/Debug/itlwm.kext

I downloaded last commit, build fwBinarry, build itlwm and still get morning in project folder (itlwm-master)

awsmct, 2020-05-16:
> I'm still getting this
> 
> [image]
> 
> 

https://github.com/zxystd/itlwm/issues/78#issuecomment-629614810

awsmct, 2020-05-16:
> just build again

Add please this line to fw_gen.sh

https://github.com/zxystd/itlwm/issues/78#issuecomment-629614810

kai2ou, 2020-05-16:
> get morning

> > If ur using the latest commit, you can get it on the projectpath/Build/Products/Debug/itlwm.kext
> 
> I downloaded last commit, build fwBinarry, build itlwm and still get morning in project folder (itlwm-master)

just build itlwm

ksh1t1z, 2020-05-16:
Go to Library/Developer/Xcode/DerivedData/itlwm.../Build/Products/Debug

ksh1t1z, 2020-05-16:
Thank to everyone here. I succeeded finally after hours.

ksh1t1z, 2020-05-16:
> > If ur using the latest commit, you can get it on the projectpath/Build/Products/Debug/itlwm.kext
> 
> I downloaded last commit, build fwBinarry, build itlwm and still get morning in project folder (itlwm-master)

https://github.com/zxystd/itlwm/issues/78#issuecomment-629625098

arch-dev, 2020-05-16:
@awsmct since problem has been solved close this issue please so we can clean issue list

closed 2020-05-16 by awsmct