OpenIntelWireless/itlwm #79: Kernel panic on Intel Dual Band AC3165

arch-dev, 2020-05-16:
Hi. I'm following this amazing project since my laptop has the AC3165 chipset installed and I've have been looking for a solution to make Intel WiFi working on my hackintosh for years.
I've built this project using latest version of xcode but when I load the kext I get kernel panic. Is it due to wrong 7265/7265D ucode firmware version?
Shortly I'll attach panic logs.

zxystd, 2020-05-16:
Seems `memcpy` function broken? I have 3165 test card and haven't encounter this problem, maybe you can try to load again?

arch-dev, 2020-05-16:
Ok I found the problem I think, it seems that changing ssid_name and ssid_pwd with mine is the cause. Should I create a new WiFi network with default credentials?

zxystd, 2020-05-16:
What is your WiFi encrypt info?

arch-dev, 2020-05-16:
WPA/WPA2-Personal

arch-dev, 2020-05-17:
I've tried disabling WPA2 in source code allowing only WPA and now it connects successfully, but after 10 seconds kernel panic is triggered. So seems that disabling WPA2 in code has changed something but is not sufficient yet.

zxystd, 2020-05-17:
I will switch to my 3165 card and test later.

arch-dev, 2020-05-17:
Ok wait for you thanks

zxystd, 2020-05-30:
Tested with 3165, no kernel panic happened...
Maybe you can try the latest code and find whether other kexts caused the kp. Thanks.

arch-dev, 2020-05-30:
Ok will try and keep you updated. Thank you for your work.

Stevemoretz, 2020-05-31:
I have 3165 no kernel panic happened for me

leopck, 2020-05-31:
I have a 3165 as well, no kernel panic for me. Worked perfectly for me for 1 week, rebooted multiple times as well. No issue.

closed 2020-06-01 by arch-dev

arch-dev, 2020-06-01:
Ok works perfectly with latest commits. Thanks.