OpenIntelWireless/itlwm #80: Tx issue

zxystd, 2020-05-16:
There is something wrong with Tx that cause firmware crash when process a large traffic upload behavior, the problem of encryption and decryption can be excluded, because it is the same when connecting to a AP without encryption. Leading to slow upload speed/random disconnect etc

vasishath, 2020-05-16:
Is there a specific size threshold above which the crash occurs or is it just random ?

kai2ou, 2020-05-16:
> Is there a specific size threshold above which the crash occurs or is it just random ?

The upload speed is slow to 5kb/s. Basically useless.

vasishath, 2020-05-16:
> > Is there a specific size threshold above which the crash occurs or is it just random ?
> 
> The upload speed is slow to 5kb/s. Basically useless.

No I am talking about the traffic upload size.

arch-dev, 2020-05-16:
Could it be due to firmware version? Maybe getting last firmware or trying another one can fix the issue you are talking about.

zxystd, 2020-05-17:
> Could it be due to firmware version? Maybe getting last firmware or trying another one can fix the issue you are talking about.

I had tried to replace to newest and oldest version of firmware, but it remains.

kai2ou, 2020-05-17:
AppleIntelWiFi.kext looks normal.

ferhatmeric, 2020-05-19:
AppleIntelWiFi.kext has no problem w/ upload speed. Max speed I get is 30Mbps both upload/download. I have 7260. 
When I switch to itlwm.kext download speed remains same but upload speed has max of 0.2Mbps

KratosAurion400, 2020-05-19:
zxystd how's your progress going regarding the upload speed issue. Also Great Work By The Way I'm Impressed Not Many People Even Bother With Trying To Get Intel Wifi Working.

balenv, 2020-05-19:
> AppleIntelWiFi.kext has no problem w/ upload speed. Max speed I get is 30Mbps both upload/download. I have 7260.
> When I switch to itlwm.kext download speed remains same but upload speed has max of 0.2Mbps

Can you share the appleIntelWiFi.kext? it has a network name and password hard coded?

tvhung83, 2020-05-20:
Oh come on, leave the guy alone, don't hijack the thread. If you have issues with compiling the kext, see other threads, there are a bunch of comments to help with that. Please keep in mind the kext is in development, think of it as POC for now, and don't use it for your daily driver.

tvhung83, 2020-05-20:
could it be Tx aggregation: https://man.openbsd.org/iwm#CAVEATS ? I have no experience at this low level, just a wild guess.

zxystd, 2020-05-20:
This problem seems to have nothing to do with 80211 stack, because when I turn off the password authentication, the process of sending data is just to remove the Ethernet frame header from the network stack and convert it to 80211 frame and send it through the PCIE, That is all, but there is still the same upload problem, confused.

zxystd, 2020-05-20:
when no encryption WiFi , the sending data sequence functions are:
`outputPacket()->_iwm_start_task()->ieee80211_encap()->iwm_tx()`

Stevemoretz, 2020-05-21:
I don't have any experience in driver writing. But I'm a professional programmer,so when I get a problem that I can't fix. I try to disable as many as things that are not related to isolate the problem.Then test it through different conditions putting logs and debugging to find out what is causing the issue.

Stevemoretz, 2020-05-21:
Posted this issue in couple of places so people come help

mehmetyukcell, 2020-05-21:
> Posted this issue in couple of places so people come help

https://www.tonymacx86.com/threads/success-working-intel-wifi-drivers-for-7265ac-on-catalina.292207/post-2125135

vasishath, 2020-05-21:
> I don't have any experience in driver writing. But I'm a professional programmer,so when I get a problem that I can't fix. I try to disable as many as things that are not related to isolate the [...]

This. If I had the time to debug this, I would actually force the driver to send lesser number of bytes until I get to know the threshold above which the firmware starts to crash. Maybe that value should give an initial pointer to the real issue. IMO, this should be a stack or a heap overflow/corruption error.

Stevemoretz, 2020-05-21:
> This. If I had the time to debug this, I would actually force the driver to send lesser number of bytes until I get to know the threshold above which the firmware starts to crash. Maybe that value [...]

I agree with this assumption if the crash happens at exactly a point, better find that point.It doesn't want much time,if it is possible to implement an automatic test an example would be start from 0 and go up log it, goto 5 goto 10 and.... Till it crashes check your logs and find the latest value if it was 250 you get that 250 is okay but 255 is too much so now start from 250 and go up one by one 251,252,253 you'll see 253 is the last possible value before crash so 254 is the number that causes the crash.Now knowing this number could've give more clue about what's going wrong.

zxystd, 2020-05-21:
The crash packet size is changeable, sometimes more than 4096, sometimes smaller than it.

[image]
[image]
[image]

I have doubted the DMA alignment, but didn't make much progress.

Stevemoretz, 2020-05-22:
The crash happens with an exception?Like a language exception error or something?Or it happens without any exception error?

vasishath, 2020-05-22:
> The crash happens with an exception?Like a language exception error or something?Or it happens without any exception error?
> 
> By language exception I meant the exceptions that are defined in the programming language you are using

This is an in-firmware crash so probably no exception.

vasishath, 2020-05-22:
> The crash packet size is changeable, sometimes more than 4096, sometimes smaller than it.
> 
> [image]
> [image]
> [image]
> 
> I have doubted the DMA alignment, but didn't make much progress.

Then there is one way but quite long though.. u may try using openbsd in a VM and passthru your wifi card. Then inside that u should be able to see how the wifi driver is operating and maybe print all headers of the packet being sent to the firmware.

Stevemoretz, 2020-05-22:
What's the reason of crash?there is some sort of runtime exception that because it's not handled,it causes the firmware to crash?It may not be an obvious exception but the crash happens because of runtime exception?

I'm just asking these to clarify I'm not sure how firmware works.

Stevemoretz, 2020-05-22:
> Then there is one way but quite long though.. u may try using openbsd in a VM and passthru your wifi card. Then inside that u should be able to see how the wifi driver is operating and maybe print [...]

This sound like a smart debugging solution for drivers, nice idea 💡

zxystd, 2020-05-22:
Firmware crash just a behavior that talks us that the packet transfer is wrong I think, because I found that sometings some routers crashed and restarted after receiving the packets sent by this driver, so amazing...

vasishath, 2020-05-22:
> Firmware crash just a behavior that talks us that the packet transfer is wrong I think, because I found that sometings some routers crashed and restarted after receiving the packets sent by this driver, so amazing...

That technically is a DoS attack lol

tvhung83, 2020-05-22:
@zxystd you mean I can use this kext to crash my neighbor routers? Sweet! 😆

tvhung83, 2020-05-22:
Jokes aside, it still sounds like TX aggregation not implemented in firmware as I mentioned in previous comment: https://man.openbsd.org/iwm#CAVEATS 
@zxystd you sure it's irrelevant?

zxystd, 2020-05-22:
> Jokes aside, it still sounds like TX aggregation not implemented in firmware as I mentioned in previous comment: https://man.openbsd.org/iwm#CAVEATS
> @zxystd you sure it's irrelevant?

I am not sure, but unless openbsd works fine. @williambj1 is trying to install openbsd in VM to verify something.

Stevemoretz, 2020-05-22:
> Firmware crash just a behavior that talks us that the packet transfer is wrong I think, because I found that sometings some routers crashed and restarted after receiving the packets sent by this driver, so amazing...

Wow how's that even possible😂what have you created zxystd?

Just kidding,it happens to me some stuff like that sometimes as well, it's probably a little tiny bug you'll find very soon and laugh about how such a small mistake caused all these issues.

zxystd, 2020-05-22:
Maybe. The download speed problem is just because I haven't zero the new allocated memory, yes, just a `bzero` function fix it.

zxystd, 2020-05-25:
[image]
[image]
sometimes the upload speed is well.

vasishath, 2020-05-25:
> [image]
> [image]
> sometimes the upload speed is well.

Great ! What was the issue ?

elnenez, 2020-05-25:
I don't know if this may be related to the TX issue, but the latest OpenBSD release (May 19, 2020, https://www.openbsd.org/67.html) fixed some problems with the iwm driver, concretely in the IEEE 802.11 wireless stack: 
  - Fixed a race condition in iwm(4) Rx interrupt handling.
  - Implemented a workaround for missing Tx completion interrupts in iwm(4) which could lead to failures when roaming to another AP.
  - Re-enabled firmware-based Tx retries at lower rates for iwm(4), reducing packet loss.
  - Fixed automatic Tx rate control issues in iwn(4), and iwm(4).

zxystd, 2020-05-26:
When the upload is well, I found that each of the maximum packet size is smaller than 1460,  as we all know, ethernet has a packet threshold which call MTU, commonly the value is 1500, payload size is 1460, it is the same as the threshold. so I assumed that it is the packet comes from network stack is MSDU packet, which is larger than MTU, but our driver doesn't support Tx aggregation now, so it crash. If I am wrong, figure me out, thanks.

vasishath, 2020-05-26:
> When the upload is well, I found that each of the maximum packet size is smaller than 1460, as we all know, ethernet has a packet threshold which call MTU, commonly the value is 1500, payload size [...]

You sound correct but isn't that supposed to be handled by TCP down at the network layer ? And in case of UDP, it attempts to send that packet and fails. It's rather strange that firmware doesn't support this

zxystd, 2020-05-26:
It's not that the firmware doesn't support it, but due to the net80211 of OpenBSD doesn't support TX aggregation, so I'm disable this feature. So now I have two options. One is to find a way to let the network stack only give me a package within the MTU size or to split it myself. The other is to implement TX aggregation, which can double the speed.

vasishath, 2020-05-26:
> It's not that the firmware doesn't support it, but due to the net80211 of OpenBSD doesn't support TX aggregation, so iwm disable this feature. So now I have two options. One is to find a way to let the network stack only give me a package within the MTU size or to split it myself. The other is to implement TX aggregation, which can double the speed.

I guess u can split the packet and loop over the remaining bytes to send it individually to the fw for now. For long term, we can look at Tx aggregation. Maybe refering to linux's 802.11 stack would help.

zxystd, 2020-05-26:
Hi there, good news, I finally fix this issue, the speed looks so nice. I will push the code and publish the release soon.

[image]

tvhung83, 2020-05-26:
Wow, that was quick, superb news! Will try it first thing tomorrow, so excited, hehe 😁

vasishath, 2020-05-26:
> Hi there, good news, I finally fix this issue, the speed looks so nice. I will push the code and publish the release soon.
> 
> [image]

Wow.. so finally what was the issue ?

zxystd, 2020-05-26:
> Wow.. so finally what was the issue ?

`IOReturn itlwm::getMaxPacketSize(UInt32 *maxSize) const {
    *maxSize = 1500;
    return kIOReturnSuccess;
}`

just override this function to tell network stack max packet size.

vasishath, 2020-05-26:
> > Wow.. so finally what was the issue ?
> 
> `IOReturn itlwm::getMaxPacketSize(UInt32 *maxSize) const { *maxSize = 1500; return kIOReturnSuccess; }`
> 
> just override this function to tell network stack max packet size.

Lol. Small things matter too much sometimes.. it should've been mandatory to override this method so the kext wouldn't compile otherwise.

tvhung83, 2020-05-26:
Actually, I was kinda hoping that TX aggregation was added, so that we can max out the speed, but that's fine, as long as it can run stable.

zxystd, 2020-05-26:
sadly, this function didn't take effect each time, sometimes it is strange like before, I will find workaround tomorrow.

tvhung83, 2020-05-27:
> sadly, this function didn't take effect each time, sometimes it is strange like before, I will find workaround tomorrow.

Take your time, buddy. Thank you for your work, it was fun, pity that I can't help anything.

zxystd, 2020-05-28:
So now the problem turns to how to config interface to limit outputpacket size below 1500. I have tried to call `setMaxTransferUnit` method, but it can't take effect, the maximun outputpacket still too large.

zxystd, 2020-05-28:
Maybe I have another way is that using `IO80211Family` to replace Ethernet interface, these configurations have been done there.

Stevemoretz, 2020-05-28:
The other kext that works AppleIntelWifi.kext doesn't use Ethernet it uses wifi.Maybe some stuff are handled via the system there if you use wifi?

rdmitry0911, 2020-05-28:
Hi guys, I have the same tx issue on my 9560 adapter with the latest git commit 7eb758ecffb550ba8d5d65f619e74f5afeaa7024. I measure throughput with iperf. Download speed is about 35 Mb/s, upload speed is about 30 Kb/s. Does this commit include fixes you wrote above?

s0ullight, 2020-05-29:
I think with commit 81 we can close this issue :)

zxystd, 2020-05-29:
Yes, thanks for all your suggest.

closed 2020-05-29 by zxystd