OpenIntelWireless/itlwm #81: macOS Catalina: start/stop routine failed

itsmattkc, 2020-05-17:
Attempting to start the kext with `kextload` results in the following Terminal message:
```
itlwmx.kext failed to load - (libkern/kext) kext (kmod) start/stop routine failed; check the system/kernel logs for errors or try kextutil(8).
```

Console.app has the following lines:
```
default	18:09:02.848645+1000	sudo	    matt : TTY=ttys000 ; PWD=/Users/matt/Desktop ; USER=root ; COMMAND=/sbin/kextload itlwmx.kext
error	18:09:02.900638+1000	kextd	Kext with invalid signature (-67050) allowed: <OSKext 0x600002534120 [0x7fff91a4c8c0]> { URL = "file:///Users/matt/Desktop/itlwmx.kext/", ID = "com.zxystd.itlwmx" }
error	18:09:03.093033+1000	kextd	Invalid signature -67050 for kext <OSKext 0x600002534120 [0x7fff91a4c8c0]> { URL = "file:///Users/matt/Desktop/itlwmx.kext/", ID = "com.zxystd.itlwmx" }
default	18:09:03.119344+1000	kernel	OSMetaClass: Kext com.zxystd.itlwmx class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
default	18:09:03.119359+1000	kernel	Kext com.zxystd.itlwmx start failed (result 0xdc00400a).
default	18:09:03.119362+1000	kernel	Kext com.zxystd.itlwmx failed to load (0xdc008017).
default	18:09:03.119364+1000	kernel	Failed to load kext com.zxystd.itlwmx (error 0xdc008017).
error	18:09:03.122780+1000	kextd	Failed to load /Users/matt/Desktop/itlwmx.kext - (libkern/kext) kext (kmod) start/stop routine failed.
```

This was compiled on macOS Catalina 10.15.4. Xcode version 11.4.

(Not sure if this is related, but my MAC address is 00:28:f8:4c:f1:30 and [AppleIntelWifi/adapter](https://github.com/AppleIntelWifi/adapter) had a bug that would fail to start because it erroneously identified the 0x00 in the first byte as an empty address)

arch-dev, 2020-05-17:
Try using kextutil instead of kextload

itsmattkc, 2020-05-17:
@arch-dev Same messages as Console.app:
```
Kext with invalid signature (-67050) allowed: <OSKext 0x7f9cb2604590 [0x7fff864558c0]> { URL = "file:///Users/matt/Desktop/itlwmx.kext/", ID = "com.zxystd.itlwmx" }
Code Signing Failure: code signature is invalid
Disabling KextAudit: SIP is off
(kernel) OSMetaClass: Kext com.zxystd.itlwmx class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
(kernel) Kext com.zxystd.itlwmx start failed (result 0xdc00400a).
(kernel) Kext com.zxystd.itlwmx failed to load (0xdc008017).
(kernel) Failed to load kext com.zxystd.itlwmx (error 0xdc008017).
Failed to load /Users/matt/Desktop/itlwmx.kext - (libkern/kext) kext (kmod) start/stop routine failed.
```

arch-dev, 2020-05-17:
Try rebooting and loading again after boot, seems that itlwm.kext is already in your kext cache

itsmattkc, 2020-05-17:
You're right, it seems to be working now. Thank you, sorry for the inconvenience.

closed 2020-05-17 by itsmattkc

duongle305, 2020-05-17:
i have same issue with load kext, this was compiled on macOS Catalina 10.15.4. Xcode version 11.4. 
[image]

arch-dev, 2020-05-17:
Don't load itl80211.kext it isn't necessary.
On the other side try rebooting your system and then type
sudo kextutil -v path_to_itlwm.kext

duongle305, 2020-05-17:
> Don't load itl80211.kext it isn't necessary.
> On the other side try rebooting your system and then type
> sudo kextutil -v path_to_itlwm.kext

kext is successfully loaded but wifi does not appear in my network
when i try load itlwmx.kext the same log this issue
[image]

itsmattkc, 2020-05-17:
@duongle305 It confused me, but it turned out all I needed to do was load `itlwm.kext` (no others) and then restart. Next boot, the kext loaded automatically and Wi-Fi worked from there.

duongle305, 2020-05-17:
My wifi appeared as ethernet but i can't connect to network, below is my log 
```
2020-05-17 21:42:42.555198+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:42:42.555219+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:42:42.555269+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:42:42.567999+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:42:42.568020+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:42:42.568072+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:42:52.563590+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:42:52.563612+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:42:52.563658+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:42:52.565696+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:42:52.565708+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:42:52.565739+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:43:02.572945+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:43:02.572953+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:43:02.572976+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:43:02.575555+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:43:02.575571+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:43:02.575616+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:43:04.856711+0700 0x36d      Default     0x0                  0      0    kernel: (itlwm) itlwm: _iwm_start_task 2543 ieee80211_encap OUTPUT_ERROR
2020-05-17 21:43:06.611891+0700 0x36d      Default     0x0                  0      0    kernel: (itlwm) itlwm: _iwm_start_task 2543 ieee80211_encap OUTPUT_ERROR
2020-05-17 21:43:09.613836+0700 0x36d      Default     0x0                  0      0    kernel: (itlwm) itlwm: _iwm_start_task 2543 ieee80211_encap OUTPUT_ERROR
2020-05-17 21:43:12.612673+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:43:12.612695+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:43:12.612741+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
2020-05-17 21:43:12.614694+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_input
2020-05-17 21:43:12.614705+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_4way_msg3
2020-05-17 21:43:12.614735+0700 0x1340     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_eapol_key_mic
```