OpenIntelWireless/itlwm #83: Intel Wireless ac 3165- Link broken and no upload speed showing as ethernet

rahulmangla19, 2020-05-18:
[image]
[image]

just to make to note to Developer and waiting For Fix.
Using Kext complied itlwm and itl80211. it working in clover and in opencore it panic spontaneously and not useful at all. 
kindly Fix the issue for broken speed ,protocol as ethernet and also support for opencore.
[image]

rahulmangla19, 2020-05-18:
[Uploading Rahul.zip…]()
kext used
[itl80211.kext.zip](https://github.com/zxystd/itlwm/files/4641713/itl80211.kext.zip)

rahulmangla19, 2020-05-18:
[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4641722/itlwm.kext.zip)
[itl80211.kext.zip](https://github.com/zxystd/itlwm/files/4641726/itl80211.kext.zip)

zxystd, 2020-05-29:
fix in 7eb758e

closed 2020-05-29 by zxystd