OpenIntelWireless/itlwm #84: Trouble w 5GHz Wi-Fi

awsmct, 2020-05-18:
Are this driver support 5GHz Wi-Fi and can SSID contain spaces? or strongly needed w/o it?

testkam2018, 2020-05-18:
i've tried with 5Ghz - no luck

awsmct, 2020-05-18:
> i've tried with 5Ghz - no luck

Whats a name of your network?

balenv, 2020-05-18:
i´ve changed the name of my 5g network to the one hardcoded ans it works,not at 5g speeds but it works

awsmct, 2020-05-18:
> i´ve changed the name of my 5g network to the one hardcoded ans it works,not at 5g speeds but it works

what does u mean by "one hardcoded"? Name consist of one word?

balenv, 2020-05-18:
the kext that im using has an ssdt and password hardcoded, where it only connect to that hardcoded network name and password. i´ve downloaded it from and issue open on this github

ferhatmeric, 2020-05-19:
No problem, both 5GHz and 2.4GHz wifi channels can be used. Max speed I get is 30Mbps.

balenv, 2020-05-19:
can you share the kext you are using?

mehmetyukcell, 2020-05-19:
> No problem, both 5GHz and 2.4GHz wifi channels can be used. Max speed I get is 30Mbps.

Ferhat Bey Merhabalar, intel 9560 wifi kart var bende bu kexti denesem önerir misiniz?

ferhatmeric, 2020-05-19:
> can you share the kext you are using?

This video has detailed description about kext file and how to load.

https://www.youtube.com/watch?v=ICpgYtzbxP8

ferhatmeric, 2020-05-19:
> > No problem, both 5GHz and 2.4GHz wifi channels can be used. Max speed I get is 30Mbps.
> 
> Ferhat Bey Merhabalar, intel 9560 wifi kart var bende bu kexti denesem önerir misiniz?

Merhaba,
9560 serisi ile ilgili karışık sonuçlar var demiş, bence deneyebilirsiniz.

Intel(R) Dual Band Wireless-AC 9560 | 8086:0030, 8086:0034 8086:0038, 8086:003c 8086:0230, 8086:0234 8086:0238, 8086:023c 8086:2030, 8086:2034 8086:4030, 8086:4034 8086:4234 | Mixed results (some are CNVi, some are PCIe) | Yes
-- | -- | -- | --

closed 2020-05-29 by zxystd