OpenIntelWireless/itlwm #85: Bluetooth connection problem for codecs AptX/AAC (SBC has no problem) while using ITLWM kext.

ferhatmeric, 2020-05-19:
Hello,

While ITLWM kext has not been loaded, intel bluetooth works perfect with all codecs I tried (AptX, AAC, SBC)

But after loading ITLWM kext, internet connection is established, but bluetooth connection drops if the codec used is AptX or AAC. And then never get connected with these 2 codecs. But SBC is exception. SBC codec has no problem with ITLWM kext. 

HW List:
808608B1: Intel Wireless 7260
808707DC: Intel Bluetooth
Lots of bluetooth headphones

Kext List:
itlwm.kext
IntelBluetoothFirmware.kext

PS: I tried to use AppleIntelWiFi.kext instead of itlwm.kext, but the result is same, there is no problem with internet connection and no KP with these kexts, but AptX and AAC codecs can not be used with itlwm/AppleIntelWiFi kexts. 

Thanks for your support
Ferhat

zxystd, 2020-05-20:
Thanks for your feedback, I will try it.

leopck, 2020-05-30:
I confirmed having similar issue with my AC-3165, tried using AppleIntelWiFi.kext/itlwm, though I don't know if it's just AptX/AAC.

ferhatmeric, 2020-05-31:
Today I tried the same with commit#83 build. 

There is no problem for SBC codec again. I can successfully connect bluetooth speaker with SBC codec while itlwm kext has been loaded.

But i can never connect bluetooth headphones with AptX or AAC codec while itlwm kext has already been loaded.

I have only one intel wifi adapter which is 7260 (808608B1) and one intel bluetooth adapter (808707DC), so I have no chance to try with different variations of wifi/bluetooth modules.

zxystd, 2020-06-13:
Is this happened again?

ferhatmeric, 2020-06-13:
For AptX/AAC codecs, bluetooth connection is NOT successful while using itlwm.kext
For SBC codec, bluetooth connection is successful while using itlwm.kext

I built and gave a try itlwm.kext by using today's (13 June) latest commit.

I think bluetooth codecs which require high bandwidth like AptX or AAC, has NOT been supported by itlwm project yet. However low bandwidth codecs like SBC is supported. 

I can provide any log files if you want.

Thanks for your great work.

zxystd, 2020-06-13:
So maybe it is due to the bandwidth conflict? If so, it's a problem that can't be solved. Maybe it's a bug of Intel card. btw, have you check the antenna?

FlasHRender, 2020-06-13:
I don't know what people are talking about but AAC works pretty well with Intel wifi. AptX seems broken irrespective of using Intel wifi or not.
[image]

ferhatmeric, 2020-06-13:
AptX codec is working perfectly without itlwm.kext as i stated at the first message.
[image]

Also AAC codec is working perfectly without itlwm.kext
[image]

But when itlwm.kext is loaded, AptX and AAC is NOT working with 7260 wifi module.
[image]

FlasHRender, 2020-06-14:
Ok AptX works sometimes. Both my KZ Earphones-HD and B&W PX are AptX HD but AAC was used for KZ and AptX for B&W PX. But yeah I'm using itlwm.kext as seen from the screenshots. In fact I'm streaming music on Tidal and Playing on AptX. So I'm sure it's not a bandwidth issue.
[image]
[image]

yaozaiyu, 2020-06-18:
我的设备是：索尼wx1000i+ax200，一直使用aptx，没有任何问题

ferhatmeric, 2020-07-20:
UPDATE:

When I select "Turn WiFi Off" via Heliport, bluetooth AptX connection successfully restores and plays music without any cut. 
After then when I select "Turn WiFi On" via Heliport, internet connection successfully restores, but AptX connection drops immediately. Again I must point out, SBC bluetooth connection has no problem with wifi powered on. Only AptX and AAC codecs have problem which uses higher bitrate than SBC.

I simulated this condition with 2.4Ghz and 5Ghz wifi connection. Both gives the same result with my 7260 intel wifi module. 

There are some people who don't face this problem with other wifi modules. 

I'm not sure but this problem occurs with 726x intel wifi modules (if there is someone who has 726x wifi module and does not have this problem please write here.)

Also a guy has posted a similar issue below with 726x wifi module below:

"7265AC Bluetooth Audio cuts out while using WiFi #191"

Maybe there is a bug with this specific wifi module, maybe hw/fw issue, I'm not sure, just a guess..

i3p9, 2020-07-22:
On the latest itlwm now, checked back to see if the issue is fixed... It's not. Wi-Fi works just fine, but when I connect a AptX headphone, the audio cuts out. So basically:

Wi-Fi is off. BT AptX audio doesn't cut out.
Wi-Fi is on. BT AptX audio cuts out.

Kexts: 
[image]

BT Stats: 
[image]

Wi-Fi Stats: 
[image]

FlasHRender, 2020-07-22:
@i3p9 please mention the wifi module you are using as on AC8260/5, AptX, AAC and SBC all work perfectly even when pushing streaming on wifi with Tidal in Master quality audio.

i3p9, 2020-07-22:
@FlasHRender edited. I'm on a 7265.

FlasHRender, 2020-07-22:
@i3p9 I think 7265 is the only device with this issue. Can you try windows using Aptx for intel bluetooth(Google search for AptX windows driver, you will find one from dell for intel 72xx and 82xx)?

i3p9, 2020-07-22:
@FlasHRender Windows has no such problems. As I use this Laptop for windows only applications I'm stuck on Windows most of the times and never had this issue.

FlasHRender, 2020-07-22:
Windows doesn't use aptx or even aac on default intel drivers. This is why I'm asking you to check with the special AptX driver for windows.

i3p9, 2020-07-22:
@FlasHRender Yes I know. That's what I meant when I said I checked via Windows, fyi I'm using this driver (which is basically the extracted version from the one you mentioned) for aptX support: https://github.com/StKob/Intel-Bluetooth-Audio

FlasHRender, 2020-07-22:
Honestly the last thing you can try is to boot Linux and check there. Intel Bluetooth kext uses Linux driver, if anything is wrong then its in the driver. Also I have noticed your HeliPort show 2.4GHz can you try with 5GHz wifi?

i3p9, 2020-07-22:
@FlasHRender Yeah I'll take a look with Linux after work, good idea. And I did try with 5Ghz before, same issue. I thought 5Ghz was gonna fix it because bt is also on 2.4ghz so there could have been be potential bandwidth/noise issue. It didn't fix the issue.

1Revenger1, 2020-07-22:
May want to look at bluetooth co-existance stuff in the driver. Linux's driver has an entire [file dedicated to coex](https://github.com/torvalds/linux/blob/cb8e59cc87201af93dfbb6c3dccc8fcad72a09c2/drivers/net/wireless/intel/iwlwifi/mvm/coex.c). Not sure how much bsd's driver had, and it'd likely be a pain to mix the two though.

FlasHRender, 2020-07-23:
@1Revenger1 The problem only exists on AC 7265. So I don't think that should be the issue.  AC8265 is handling 4 BT devices all at one along with WiFi streaming.

invalidsudo, 2020-07-31:
I have a 7260AC in my ThinkPad T440p running macOS Catalina 10.5.6. Having the same issue here with my 1st gen. AirPods, which use the AAC codec. The Bluetooth range is reduced to the immediate area surrounding the laptop, so signal quality becomes poor or completely gets dropped once I move further away while using the AirPods. 

This issue does not occur when using the AirPods in Windows or Linux however. It also does not occur when uninstalling itlwm.kext and using a Wi-Fi only adapter alongside IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext instead. This means that this is an issue with itlwm.kext. Changing router settings as suggested in other issues did not help.

Also, I tried the solution [here](https://gist.github.com/xbeta/e60fddd012e95202bbbe), but it did not fix the issue.

invalidsudo, 2020-07-31:
I have a 7260AC in my ThinkPad T440p running macOS Catalina 10.5.6. Having the same issue here with my 1st gen. AirPods, which use the AAC codec. The Bluetooth range is reduced to the immediate area surrounding the laptop, so signal quality becomes poor or completely gets dropped once I move further away while using the AirPods. 

This issue does not occur when using the AirPods in Windows or Linux however. It also does not occur when uninstalling itlwm.kext and using a Wi-Fi only USB adapter alongside IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext instead. This means that this is an issue with itlwm.kext. Changing router settings as suggested in other issues did not help.

Also, I tried the solution [here](https://gist.github.com/xbeta/e60fddd012e95202bbbe), but it did not fix the issue.

invalidsudo, 2020-07-31:
I have a 7260AC in my ThinkPad T440p running macOS Catalina 10.15.6. Having the same issue here with my 1st gen. AirPods, which use the AAC codec. The Bluetooth range is reduced to the immediate area surrounding the laptop, so signal quality becomes poor or completely gets dropped once I move further away while using the AirPods. 

This issue does not occur when using the AirPods in Windows or Linux however. It also does not occur when uninstalling itlwm.kext and using a Wi-Fi only USB adapter alongside IntelBluetoothFirmware.kext and IntelBluetoothInjector.kext instead. This means that this is an issue with itlwm.kext. Changing router settings as suggested in other issues did not help.

Also, I tried the solution [here](https://gist.github.com/xbeta/e60fddd012e95202bbbe), but it did not fix the issue.

FlasHRender, 2020-07-31:
@invalidsudo Hey the bug is either in the firmware or the device both of which can't be fixed as 7260/5 is the only device with the bug. As all other devices are using the same method and are working fine, this means that there is no issue with itlwm kext.

zxystd, 2020-09-09:
I think it is a card model related issue, which I found it is happened on AC7260(test on my ax201 and ac9260 works fine). So, hmmm, I think I don't need to fix it. Thanks.

ferhatmeric, 2020-09-09:
Understood. I've already bought external bluetooth dongle and it's working fine with intel 7260 wifi. 
Thanks for your effort.

closed 2020-09-09 by ferhatmeric

Doregon, 2020-12-01:
@zxystd only the fact that this isn't just an issue with that card. The AC 3165 has the exact same problem, and the N 7260.