OpenIntelWireless/itlwm #88: AC 9462 doesn't work

destr0y, 2020-05-24:
I put "itlwm.kext" and "itl80211.kext" to Clover\Kexts\Other, change SSID to "ssdt" and password to "zxyssdt112233" and rebooted the system.
But... nothing happend...
WiFi icon didn't appear in menu bar, in Preferences->Network WiFi or ethernet didn't appear too.
What did I do wrong or what should I do?
I use Mac OS 10.15.5 and my WIFi card is AC 9462

zxystd, 2020-05-25:
ProductId? you can get it in hackintool.

destr0y, 2020-05-26:
I don't know where I can get productid and I didn't find productid in hackintool, but I found my card in PCIe.
Vendor: 0x8086, Device: 0x02F0, SubVendor: 0x8086, SubDevice: 0x02A4, Device Name: Wireless-AC 9462.
If it isn't that you want please tell me how can I find productid

zxystd, 2020-05-28:
0x02f0 is the product id, it is a gen2 device, will not support in `itlwm`, but maybe will support in `itlwmx` which is make to 2000 series devece, after I debug the `itlwmx`, maybe you can try again. Thanks.

KratosAurion400, 2020-05-29:
Great Work Glad The TX issue is finally solved..do you plan on releasing a guide in the near future?

zxystd, 2020-05-29:
I will release the binary file, but no install guide, just put it in bootloader kext directory and it will work, because the work isn't done, we should have a client to manage Wifi connection, and the speed can be more improve.

KratosAurion400, 2020-05-29:
Ok thanks

KratosAurion400, 2020-05-29:
So Do I need to compile a separate next for my Intel AX 200 Wifi Card?

agassecond, 2020-07-06:
Any updates on gen2? Having same 02f0 model

mrkaratechamp, 2020-07-11:
hi,  I have exact same issue as in  the  #168 post, same logs  (invoke,  free,free,free...),  but my card not work,  
I  have matebook 13 2020, so I  think it's same problem,  2 gen chipset right?,  I already try  itlwm fist, but when  it's not work, I read more and then compile the x kext to use,  but  no  luck, this issue still  pressent? or maybe I miss something?
thanks!!

nekiro, 2020-07-26:
Yes, looks like the issue is still present, because I just noticed it does not work and I have the exact same device.

zxystd, 2020-07-27:
sorry, I am going to open a new issue to discuss about this support. please turn to #214

closed 2020-07-27 by zxystd