OpenIntelWireless/itlwm #90: 安装之后系统会经常重启

chlp1055378613, 2020-05-28:
panic(cpu 0 caller 0xffffff7f980a0231): nvme: "Fatal error occurred. CSTS=0x1 US[1]=0x0 US[0]=0x5c VID=0x144d DID=0xa808
. FW Revision=EXD7201Q\n"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/IONVMeFamily/IONVMeFamily-470.100.17/IONVMeController.cpp:5320
Backtrace (CPU 0), Frame : Return Address
0xffffff82f2b3b9e0 : 0xffffff8015b215cd 
0xffffff82f2b3ba30 : 0xffffff8015c5a3c5 
0xffffff82f2b3ba70 : 0xffffff8015c4bf7e 
0xffffff82f2b3bac0 : 0xffffff8015ac7a40 
0xffffff82f2b3bae0 : 0xffffff8015b20c97 
0xffffff82f2b3bbe0 : 0xffffff8015b21087 
0xffffff82f2b3bc30 : 0xffffff80162c2c7c 
0xffffff82f2b3bca0 : 0xffffff7f980a0231 
0xffffff82f2b3bcc0 : 0xffffff7f9808b362 
0xffffff82f2b3be20 : 0xffffff8016233949 
0xffffff82f2b3be90 : 0xffffff8016233869 
0xffffff82f2b3bec0 : 0xffffff8015b63545 
0xffffff82f2b3bf40 : 0xffffff8015b63071 
0xffffff82f2b3bfa0 : 0xffffff8015ac713e 
      Kernel Extensions in backtrace:
         com.apple.iokit.IONVMeFamily(2.1)[3DB9BFC6-422F-39A2-A445-AB2732CE5BFC]@0xffffff7f9807d000->0xffffff7f980bffff
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[5EF51D44-9FC7-357F-9FB7-3468FD748382]@0xffffff7f967f9000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[1B1F3BBB-9212-3CF9-94F8-8FEF0D3ACEC4]@0xffffff7f968b3000
            dependency: com.apple.driver.AppleEFINVRAM(2.1)[6B3E9057-454A-3220-8AE3-424808CDA9E5]@0xffffff7f96abb000
            dependency: com.apple.iokit.IOStorageFamily(2.1)[425BC668-32EC-368C-B4EB-CF8510846BEE]@0xffffff7f96465000
            dependency: com.apple.iokit.IOReportFamily(47)[337B24B2-0261-3521-8D1B-23387FC1D657]@0xffffff7f9649f000

BSD process name corresponding to current thread: kernel_task
Boot args: igfxcflbklt=1 darkwake=0 gfxrst=1 dart=0 -cdfon -wegnoegpu 

Mac OS version:
19E287

Kernel version:
Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64
Kernel UUID: AB0AA7EE-3D03-3C21-91AD-5719D79D7AF6
Kernel slide:     0x0000000015800000
Kernel text base: 0xffffff8015a00000
__HIB  text base: 0xffffff8015900000
System model name: MacBookPro15,3 (Mac-1E7E29AD0135F9BC)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 107539274500
last loaded kext at 10780727163: com.zhong.!A!IWiFi	1 (addr 0xffffff7f991d4000, size 1019904)
last unloaded kext at 8667421256: com.cisco.kext.acsock	4.6.0 (addr 0xffffff7f99023000, size 217088)
loaded kexts:

zxystd, 2020-05-28:
You are not using this kext and will not support here, thanks.

chlp1055378613, 2020-05-28:
sorry

closed 2020-05-28 by zxystd