OpenIntelWireless/itlwm #91: Request a wifi ui selector todo

Stevemoretz, 2020-05-29:
Hi thanks for all the great hard work,you've said it before that you rather do the ui,after the crashes and speed problems are gone, with your latest commit you have solved the above issues, so now we'd love to see the ui selector replace with the hard-coded said and password.That would be very nice and takes this kext one more big step to the release stage.

williambj1, 2020-05-29:
WIP :p

Klingens13, 2020-06-08:
> WIP :p
Not to be rude, but can we get an very rough ETA based on priority?

closed 2020-06-13 by zxystd

Stevemoretz, 2020-06-13:
@zxystd sorry to ask, but you closed it because it has been implemented,or because it will be implemented?

zxystd, 2020-06-13:
It is WIP, you can see it in my another repo call HeliPort. Any related issus can turn to there. thanks.

https://github.com/zxystd/HeliPort