OpenIntelWireless/itlwm #92: Low throughput issue

rdmitry0911, 2020-05-29:
Hi guys, first of all, thank you for the efforts you've invested in this project. With the latest commit wifi became usable for the real life. However, the throughput is still much less, then the wifi card can provide. To test it I'm using iperf. With the latest driver under 10.15.5 I have these results:

Client connecting to 172.16.66.211, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  6] local 172.16.66.219 port 62454 connected with 172.16.66.211 port 5001
[ ID] Interval       Transfer     Bandwidth
[  6]  0.0-10.0 sec  7.62 MBytes  6.39 Mbits/sec
[  4]  0.0-10.9 sec  18.5 MBytes  14.3 Mbits/sec

In Ubuntu 20.04 I have this:

Client connecting to 172.16.66.211, TCP port 5001
TCP window size:  129 KByte (default)
------------------------------------------------------------
[  6] local 172.16.66.219 port 62593 connected with 172.16.66.211 port 5001
[ ID] Interval       Transfer     Bandwidth
[  6]  0.0-10.0 sec  55.1 MBytes  46.2 Mbits/sec
[  4]  0.0-10.3 sec  51.1 MBytes  41.7 Mbits/sec

So, the card can provide 3x faster download and 7x faster upload. 
Are there any plans to improve performance or you have now other priorities?

zxystd, 2020-05-29:
This is not a issue, just a enhancement I think.
Now it only max support 11n, 20mhz band width WiFi, too many things need to do. such as Tx agg, tso offload, 40mhz support, 11ac support even 11ax and so on. Linux has full support of these.
By the way, are you using wpa2/ccmp WiFi? and what's your card model?

rdmitry0911, 2020-05-29:
> This is not a issue, just a enhancement I think.
> Now it only max support 11n, 20mhz band width WiFi, too many things need to do. such as Tx agg, tso offload, 40mhz support, 11ac support even 11ax [...]

Yes I'm using wpa2/ccmp and I have LG Gram 17 with 0x8086,0x9df0 cannon point-lp cnvi wireless-ac adapter (9560 series) As for the environment, I have old UniFi AP configured for 11n with 20mhz:

RADIO (11N/B/G)
Channel	Auto (11) HT20
Transmit Power	17 dBm / 20 dBm (EIRP)
Tx Pkts/Bytes	129,706,391 / 21.4 GB
Rx Pkts/Bytes	395,892,413 / 198 GB
Tx Retry/Dropped	17.4% / 0.0%
Rx Retry/Dropped	0.0% / 0.1%
Ch. Util. (Busy/Rx/Tx)	5% / 1% / 1%
# Users	1
# Guests	0

So, it should be fine with what is supported in the driver at the moment.

closed 2020-06-13 by zxystd