OpenIntelWireless/itlwm #94: Cannot connect to a network with a single letter name

igorkulman, 2020-05-30:
I am running two networks from my home router,  `K` and `K-guest`. Using the latest source code that allows setting the networks in `Info.plist` on my AC7260 it never connects to `K`.

If I add both networks to the config, `K` first, it connects to `K-guest`. If I add just `K` to the config, it does not connect at all.

So maybe it is related to the short SSID? Both networks run from the same router, have same security settings, just different passwords (numeric vs alphanumeric).

When connected to `K-guest` the kext works quite well, getting very good speeds.

zxystd, 2020-05-30:
log show --last boot | grep itlwm >itlwm_log.txt

run the command and send me the log, or I don't know what happened.

igorkulman, 2020-05-30:
Here you go. `log1.txt` is with just `K` in the config that never connects. `log2.txt` is with bot networks, connecting to `K-guest`.

[log1.txt](https://github.com/zxystd/itlwm/files/4705535/log1.txt)
[log2.txt](https://github.com/zxystd/itlwm/files/4705536/log2.txt)

zxystd, 2020-05-30:
ok, got it. And I find the error is caused by the `trim` method length calculation mistake, will update soon.Thanks.

igorkulman, 2020-05-30:
Glad to have helped. Keep up the good work I rally appreciate this project.

zxystd, 2020-05-30:
[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4705763/itlwm.kext.zip)
help me to check if I have fix it.

igorkulman, 2020-05-30:
> [itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4705763/itlwm.kext.zip)
> help me to check if I have fix it.

When I put just `K` to the config then it works fine, connect to the network, so the that is fixed. Thanks.

But when I put both `K` and `K-guest` then it connects to `K-guest` instead o `K`. `K` is first in the config as `Wifi_1`. 
[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4705795/itlwm_log.txt)

ksh1t1z, 2020-05-30:
> [itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4705763/itlwm.kext.zip)
> help me to check if I have fix it.

Hey, so I tried this kext you built after failing to load the one I built.
I'm getting the same error while loading it.
I'm on 10.14.6.

MacBook-Pro:Downloads bestware$ sudo kextload -v itlwm.kext
Password:
Requesting load of /Users/bestware/Downloads/itlwm.kext.
/Users/bestware/Downloads/itlwm.kext failed to load - (libkern/kext) link error; check the system/kernel logs for errors or try kextutil(8).
MacBook-Pro:Downloads bestware$ sudo kextutil -v itlwm.kext
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext with invalid signatured (-67050) allowed: <OSKext 0x7ffb6ec02e50 [0x7fffaeee98e0]> { URL = "file:///Users/bestware/Downloads/itlwm.kext/", ID = "com.zxystd.itlwm" }
Code Signing Failure: code signature is invalid
/Users/bestware/Downloads/itlwm.kext appears to be loadable (not including linkage for on-disk libraries).
Loading /Users/bestware/Downloads/itlwm.kext.
Disabling KextAudit: SIP is off
KextAudit initialized: audit=F
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV13IOEventSource' for vtable '__ZTV11IOTaskQueue' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV19IOEthernetInterface' for vtable '__ZTV15itlwm_interface' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV15pci_intr_handle' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV20IOEthernetController' for vtable '__ZTV5itlwm' is out of date. Make sure your kext has been built against the correct headers.
(kernel) kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV8CTimeout' is out of date. Make sure your kext has been built against the correct headers.
(kernel) Can't load kext com.zxystd.itlwm - link failed.
(kernel) Failed to load executable for kext com.zxystd.itlwm.
(kernel) Kext com.zxystd.itlwm failed to load (0xdc008016).
(kernel) Failed to load kext com.zxystd.itlwm (error 0xdc008016).
Failed to load /Users/bestware/Downloads/itlwm.kext - (libkern/kext) link error.
Failed to load /Users/bestware/Downloads/itlwm.kext - (libkern/kext) link error.
Check library declarations for your kext with kextlibs(8).

igorkulman, 2020-05-30:
> Hey, so I tried this kext you built after failing to load the one I built.
> I'm getting the same error while loading it.
> I'm on 10.14.6.
>
> MacBook-Pro:Downloads bestware$ sudo kextload -v itlwm.kext [...]

Disable SIP

zxystd, 2020-05-30:
Should be notice that the order in info.plist didn't equal to AP selection priority. The driver will calculate AP's scores for selection:
crypto: wpa2 > wpa1 > wep > open
band: 5 GHz > 2 GHz
rssi: rssi1 > rssi2

For a better experience, maybe we can wait for the Heliport project. Thanks.

zxystd, 2020-05-30:
> > [itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4705763/itlwm.kext.zip)
> > help me to check if I have fix it.
> 

I compile it use 10.15 sdk, so you can't use in 10.14, maybe you can compile by self or wait for my publish?

igorkulman, 2020-05-30:
> Should be notice that the order in info.plist didn't equal to AP selection priority. The driver will calculate AP's scores for selection:
> crypto: wpa2 > wpa1 > wep > open
> band: 5 GHz > 2 GHz
> rssi: rssi1 > rssi2
> 
> For a better experience, maybe we can wait for the Heliport project(WiFi management client). Thanks.

How often is the calculation done? For example if I live to another room will it switch to a stronger network if available? Or only if the already connected network becomes unavailable?

zxystd, 2020-05-30:
> How often is the calculation done? For example if I live to another room will it switch to a stronger network if available? Or only if the already connected network becomes unavailable?

I understand what you said, this functionality call ROAM, should be work with background scan, but now background scan have problems(you can see it in Project Bugfix), Will support soon I think.

closed 2020-05-30 by zxystd