OpenIntelWireless/itlwm #95: Matebook X Pro2020，ac9560无法使用

kdzmwl, 2020-05-30:
[image]
执行加载命令后如图所示，也没有出现WiFi、以太网等设置。

lsvih, 2020-05-30:
没记错的话这个是 gen2 的卡，等作者的 itlwmx 吧。。#88 #93

zxystd, 2020-05-31:
productId? 用hacintool在pci那一栏可以看到

tsingui, 2020-06-01:
Nuc8i7beh AC9560 doesn't work

weidongbcn, 2020-06-01:
你放在什么位置？我直接放到clover的kext-other里头，重启就可以用了。是2个文件。

kdzmwl, 2020-06-02:
> 你放在什么位置？我直接放到clover的kext-other里头，重启就可以用了。是2个文件。

我直接放在桌面上，用terminal加载的。引导是oc

weidongbcn, 2020-06-02:
[image]
[image]
我用oc启动，console中调用，也可以用。

kdzmwl, 2020-06-02:
> [image]
> [image]
> 我用oc启动，console中调用，也可以用。

需要将驱动放在特定位置吗？我直接放在了桌面。
另：请问您的笔记本型号和我的是同一款吗？

weidongbcn, 2020-06-03:
[image]
[image]

你的设备属性表有弄好吗？

kdzmwl, 2020-06-03:
> [image]
> [image]
> 
> 你的设备属性表有弄好吗？

新人，对config中项目的含义不是特别理解。我只需要用命令加载，也需要在config中添加这些项目吗？如果方便，是否可以添加一下QQ详谈

zxystd, 2020-06-03:
如果你不打算回答我上面提的问题的话，那这个issue可以关了。

kdzmwl, 2020-06-03:
> 如果你不打算回答我上面提的问题的话，那这个issue可以关了。

[image]
非常抱歉！没有及时回复您的问题。网卡应该是ac9560，但是在hackintools里面显示是上图的样子。

zxystd, 2020-06-03:
ProductId 是0x02F0，这个id跟ax201一致，属于二代设备，将会在另一个驱动itlwmx里支持。

closed 2020-06-05 by kdzmwl