OpenIntelWireless/itlwm #96: Terminal-based Connection

gplast, 2020-05-30:
Hello @zxystd,

Thank you again for this great work!
After your last commit i can also confirm that the drivers are working great(no issues :) )  on my Intel8265 wifi card (both upload and download).

I was waiting until we were able to use the kextunload command to tell you that i've created a command based connection script in order to enable users to connect on different Wifi Access Points by giving the ssid and the password from the terminal. I'm sharing this here first because this is the original repository for this awesome drivers :) 

Attached you can find the script along with the updated itlwm.kext where i renamed the ssid and the password in order to find them through the script. 

First **unzip AutoConnect.zip**, **open AutoConnect** and move both the **connect.command and the itlwm.kext.zip** on the **Desktop**. Then open terminal and run:
- cd Desktop
- sudo chmod +x connect.command
- Here you have two options, either run ./connect.command or close the terminal and double click on the connect.command. If the default action with .command files is to run them on the terminal, this will open the terminal and execute the commands. Follow the instructions printed on the terminal.

**CAUTION: At the end of the script i'm killing all the terminals, so if you don't want to close all terminals, comment out the last line.**

Thank you again and i hope that you will find this script useful :) 

[AutoConnect.zip](https://github.com/zxystd/itlwm/files/4705987/AutoConnect.zip)

williambj1, 2020-05-30:
1. It says macOS can't identify whether it's malware and blocks it from executing, I had to right-click to execute it.

2. Why not replace `killall Terminal` with `exit 0`? I'm using the built-in terminal in VS Code and it won't get killed ;p

3. The client tool is currently WIP, stay tuned!

gplast, 2020-05-30:
I was using exit 0, but if i was double clicking the .command file, the terminal was staying opened ;p I think that exit 0 is better solution though :)

Stevemoretz, 2020-05-31:
This is very nice and all but I have %&$#/ in my password causing error.

Stevemoretz, 2020-05-31:
I changed your sed with plistbuddy and yet my mobile Hotspot only works which has only digits in password.wifi doesn't work even though the password is in plist.is there a bug with this kind of passwords with kext?

Stevemoretz, 2020-05-31:
I created a separate issue for this problem

zxystd, 2020-07-17:
Now we have HeliPort, appreciate to your works.

closed 2020-07-17 by zxystd