OpenIntelWireless/itlwm #97: intel 9560ac appears as Ethernet 2 on Network preferences and it has connect internet, interestingly

mehmetyukcell, 2020-05-30:
Built new itlwmx kext, loaded  with
cd desktop 
sudo chown -R root:wheel itlwm.kext
sudo kextload itlwm.kext

successfully and my 9560ac appears as Ethernet 2 on Network preferences. Also, on Hackintool it shows in the attached photo. However, it has connected intetnet with USB WİFİ Asus nano ac53. demount USB WİFİ but still connect internet. Surprise.... interesting..Also, builted kext 488,4 mb.

[image]
[image]

zxystd, 2020-05-31:
4884mb... how did you do that?

mehmetyukcell, 2020-06-01:
> 4884mb... how did you do that?

with Xcode build kext and product iltwm.kext :)I did not understand too. Also, pci ac9560 and cnvi ac9560 is there a difference between the two?

fanvantan, 2020-06-08:
I have nearly same issue with 9560AC (device ID 9DF) in Mojave. 
After loading kext, network shows Ethernet 2 (not WI-FI so there is no SSID to connect). And Ethernet 2 doesn't receive DHCP, apparently.
Hackintool shows Cannon Point-LP CNVi [Wireless-AC] like @mehmetyukcell

closed 2020-06-13 by zxystd