OpenIntelWireless/itlwm #100: Doesn't work on my router with hidden ssid with special character passwords

Stevemoretz, 2020-05-31:
Hi there is a bug, my Hotspot phone password is with digits 64373737

My password ssid is hidden and also has a hard password containing @$&#/%
There is no internet on my router but my Hotspot phone works just fine.

Stevemoretz, 2020-06-13:
Hi I forgot to say I tested it via
https://github.com/zxystd/itlwm/issues/96
This tool it gives an ability to edit plist and put ssid and password in there and then it works with regular names but for hidden + special characters it fails. Any idea? Is this considered to be solved?

zxystd, 2020-06-13:
You should not call it as "issue", it is a new feature, not a bug. Hidden ssid auth is different from common scan, it should broadcast ssid when scanning. just wait for the new feature. thanks.

Stevemoretz, 2020-06-13:
Oh I didn't know that I thought it was a small problem with the code.Thanks waiting for it.

zxystd, 2020-07-08:
now itlwm can connect to hidden ssid network(use heliport and join to network). itlwmx also can work, but crash after association, need to be fixed.

Stevemoretz, 2020-07-08:
Thank you zxystd you've done some neat job.Well done honestly you're the best. How far from a release kext you'd think we are? Seems pretty close to me.

Stevemoretz, 2020-09-22:
Any updates on this?

Stevemoretz, 2020-10-25:
Just checked with airport version still not working

Stevemoretz, 2020-10-25:
I do confirm that heliport works but only for a few seconds then crashes again and again :/

williambj1, 2021-02-25:
Feel free to request a reopen and continue the discussion with the debug logs provided.

closed 2021-02-25 by williambj1

Stevemoretz, 2021-02-25:
Is it implemented now?
In which version?

williambj1, 2021-02-25:
> Is it implemented now?
> In which version?

Have you tried the latest alpha?

Stevemoretz, 2021-02-25:
Not yet, so it finally happened, very well done!

The work is done in this repo is really admiring!
Keep up the great work.
Thanks for this nice feature.

MohammedBadr20s, 2021-11-18:
@Stevemoretz  I'm using version Number v2.0.0 stable and can't connect to hidden ssid isn't it implemented in the latest version yet ?